opentrack/opentrack #652: Freetrack vs OpenTrack?

Luc45, 2017-07-11:
Hello everyone!

First of all, thanks a lot for supporting this project, head tracking is awesome!

Secondly, I'm a old user of Free-Track.net, it works nice, but the latest release was in 2008.

I'm a bit lost about opentrack, can you please explain what are the main differences between opentrack and freetrack, just so I can catch up?

I'm sure some free-track users might have the same question, and since free-track is also an open project and has no profit in mind, I'll post this question on free-track forum just so everybody can keep informed about the latest open source projects in head tracking, still in development.

sniperwolfpk5, 2017-07-12:
The basic theme is same. If you are free track user then you have to use point-tracker option in opentrack. Open track is not just limited to point track(tracking with hat or led clip) It has a more tracking options some of them are face tracking (you just need web came thats it), Hatire tracking, PI tracking (To use your mobile phone's gyro scope for tracking), Some VR tracking, Intel tracker etc. There are a lot more I just named some of them which I know about. There are multiple output protocols which is targeted for different applications. 

**For free track user** you have to select point-tracker from tracker dropdown list and protocol as FreeTrack. Thats it. You can tune your axis just like you do in freetrack and you can setup your clip or hat in point-tracker options. I hope this will help you.

Len62, 2017-07-12:
I don't think that face tracking is available in OT at this time. Only briefly used freetrack a while back when I discovered OT, have been with it ever since. Some basic info available in the wiki. I use a 3 pt clip and modified PS3 eye with OT, and it works great.

sthalik, 2017-07-15:
I'd personally avoid going into "which is best" territory. Some users are freetrack holdouts and "convincing" them is poor form.

- opentrack supports multiple output protocols as well. There's quite a few of them now.
- opentrack is actively maintained. There aren't known issues lingering for years.
- I don't know the state of filtering in freetrack. Can someone make a fair comparison?
- freetrack has few options we haven't implemented yet, e.g. rotation decrease on zoom.
- freetrack has support for LED models other than 3-point.

closed 2017-12-21 by sthalik