opentrack/opentrack #653: 'opentrack.exe' occasionally closes itself when restarting tracking

OCTIAT, 2017-07-13:
Hello,
I've got a problem with this program. I don't know if this issue has been posted around here before but I'd be grateful if it would get fixed or if someone could help me fix it (in case the problem is on my end).

When I move my head too far in the yaw or pitch axis, for example, the tracking often goes out of sync, all the axes become inverted I think and recentering doesn't really help.  Restarting the tracking fixes this issue, so I assigned a button shortcut (F10) for that but about 25% or so of the time restarting the tracking using that shortcut causes 'opentrack.exe' to close itself and the tracking gets stopped.

**Input:** Pointracker 1.1
**Output:** freetrack 2.0 Enhanced
**Filter:** Accela

I have 'dynamic pose resolution' on and if you need any more details, please ask away. The aforementioned issue is not really making the program 'broken' but I would really appreciate if you could help me fix it.

Many thanks in advance, especially to those who have been helping to develop this amazing program.

Len62, 2017-07-13:
If you are using a 3 pt clip then dynamic pose should be unchecked, it's for hat use. I just use one shortcut to "toggle" the tracking, as opposed to separate ones for stop/start. Which version of OT ?

OCTIAT, 2017-07-13:
@Len62 Hi, thanks for the comment. I use the latest OT version (2.3.7). I actually use a hat with 3 led points so the dynamic is probably not the cause, although I didn't know you should disable it with a clip. I just ditched 'Restart tracking' for 'Toggle tracking' now and I'll see how it goes. I took a look again at the settings and I think one of the issues I experience

> When I move my head too far in the yaw or pitch axis, for example, the tracking often goes out of sync, all the axes become inverted I think and recentering doesn't really help.

is possibly related to 'Relative translation'. I had it 'unticked' so far, should I turn this setting on?
Anyway, I think I'll test that and post or edit in the outcome.

Len62, 2017-07-14:
Download this pre-release version: https://www.dropbox.com/sh/823dd4wyhixv5a7/AABoerfMr8hBhiyDqmj3eXLFa/opentrack-test-20170707_01.7z?dl=0

Fixes the issue causing the tracking to go out of sync.

OCTIAT, 2017-07-14:
@Len62 I just tried this version and the sync problem is still present. The 'opentrack.exe' still seems to disappear about one-third of the time when **toggling**/restarting and sometimes even randomly (though it used to be that way with the previous version too). I tried to play with the relative translation setting but to no avail. It seems that whenever I move my head (accidentally or not) over 90 degrees to the right or left, the pitch goes all the way up, the yaw axis gets inverted and other axes don't seem to work like they should as well. I had a different problem which I forgot to mention in the main post that did get fixed in this pre-release version though. The issue was with the pitch axis, which caused the camera in my game to go to the right whenever I tried to look up over 40 degrees or so.

Len62, 2017-07-14:
Almost starting to sound like a calibration issue. Have you done the calibration? Is it possible for you to post a short video that shows what is happening? Which OS are you running, and are you running OT at the admin level?

OCTIAT, 2017-07-15:
@Len62 I'm running Windows 10 (Creators Update). I just set it to open as administrator and I think the crashes when restarting the tracking are now finally gone! 👍 Really thanks for that.

[These](http://i.imgur.com/usdIFw0.png) are my calibration settings. I tried to calibrate it like in the YouTube video linked in the wiki, so the yellow cross is around my nose. I also tried to measure the dimensions of my cap, hopefully those values are right.

I've played a little with the 'dynamic pose timeout' setting and it seems that increasing the time to 2750ms helps a bit with the sync issue. Now only when I'm doing kinda extreme movements the problem reappears. Completely disabling 'dynamic pose resolution', however, has an unwanted side effect which causes the camera to rotate to high pitch every time I yaw over ~45 degrees.
In the [FaceTrackNoIR's PointTracker plugin](http://ftnoirpt.sourceforge.net/) there's a 'Reset' button next to the 'dynamic pose resolution' that should probably fix that sync issue I'm experiencing, though I doubt such a feature would prove useful even if it was implemented in opentrack, unless there would be a hotkey assignment for that.

I uploaded [this](https://youtu.be/qU6blxWhhI0) video where I rotate yaw ~150 degrees to the right, which causes the yellow cross to go crazy and as you can see recentering doesn't help, only restarting the tracking does. In game, it seems to happen a bit more easily than doing such a rotate, especially when I roll my head or rotate it towards one of my roommates or the trash can, even when using the 'Zero' or 'Toggle' hotkeys beforehand.

Anyway, it's not a really big deal any longer—I can just restart the tracking whenever the issue reoccurs because the crashing is now gone for good I think. Thanks for the help again 👍

sthalik, 2017-07-15:
@OCTIAT,

> I can just restart the tracking whenever the issue reoccurs because the crashing is now gone for good I think. 

I recommend covering a single LED for the timeout's duration.

The crashes are related to the PS3 Eye camera driver and the USB controller. Plugging it somewhere else helped me. My conjecture here is that if the camera can't close itself quickly, opening it again will crash it.

Len62, 2017-07-15:
@OCTIAT In the video when tracking starts your leds are almost at the top of the frame. Try and adjust your camera that so you get a more centered view. Also, center should be around the neck area. What resolution do you have the camera set at in OT? Are you using IR led's?

OCTIAT, 2017-07-15:
@sthalik What do you mean by "covering a single LED for the timeout's duration"? Do you suggest covering the top LED (for example) when there's a need to do an inconvenient movement?
Actually the program doesn't crash anymore after I had set the client to run as administrator, but I'll try to keep that in mind in case I experience crashing again in the future. Also, thank you for developing this program, it's really good!

@Len62 Camera resolution is 640x480 (75HZ and FOV of 75 degrees). Yes, I'm using IR LEDs. I have followed your suggestion, now it looks like [this](http://i.imgur.com/PbILMK3.png) when I'm in a typical position. However, the yellow cross still seems to get lost when rotating ~135 degrees to the left or right.

To be honest, I don't mind it anymore now that the program doesn't crash, I can just restart the tracking with the ease of a hotkey. Though I also don't mind to continue to comment to provide you with more details in case you haven't heard of this before and it's not a usual occurrence.

sthalik, 2017-07-15:
@OCTIAT if you cover a LED for the timeout's duration, the timeout will trigger. It's less faulty than resetting the camera.

> Actually the program doesn't crash anymore after I had set the client to run as administrator

It's random chance it doesn't crash now. Running as admin doesn't matter here. It crashed for me with UAC off.

Len62, 2017-07-15:
@OCTIAT I don't know if this matters, or you've already done this, but when you remove the IR filter it changes the focus of the camera. If you turn the lens to the left (red mark) and then very slowly rotate it you can get it to stop between the two marks, sort of a makeshift focus point. Do any of your led's overlap, or go off-screen, while moving your head?

OCTIAT, 2017-07-15:
@sthalik I just tried to cover one LED and I think it does prevent the yellow cross from going nuts. Interesting. Thanks.
@Len62 I haven't done that actually. One or more LEDs goes off-screen while moving my head now with the lens not set on either dot. Increasing the threshold seems to fix that, however. Was my threshold too low? I'm not so sure. If I switch to the 75 FOV mode and increase the threshold slider by 10+/-, it tracks the top led better, though it also causes the LEDs to flash more often, especially if I'm my head is not very close to screen (like when I lean on my chair). It seems that the previous threshold also had that quirk, but as I said, not very often. The sync problem still persists even with higher/lower threshold. I really think it's related to the 'dynamic pose resolution' setting. If I set the time to higher than 500ms, it gives me more time to return to the previous position, but if I set it too high (5000ms), it can take a little time to reset the yellow cross correctly if my rotation is quick. It seems like turning over 90 degrees yaw to the right or left is enough to trigger the sync issue, instead of 135/150 degrees like I said before.
Maybe it's not an issue but just how PointTracker works. If so, I apologize in advance for having you read all this text.

Len62, 2017-07-15:
In OT try setting the point tracker field of view to 56 deg. In the threshold settings make the min 5 and the max around 14-15. If the LED flashes it could be due to the min not being set low enough. I don't mind at all reading your posts, I'm by far not an OT expert, but I try and help where I can. I use a 3 pt clip w/PS3 cam and that somewhat restricts my knowledge of other options in OT.

sthalik, 2017-07-16:
@OCTIAT if it flashes, either your threshold is too high or camera exposure and gain are too low.

OCTIAT, 2017-07-16:
@sthalik I had the exposure&gain sliders set to zero in the CL-Eye driver because I had read somewhere that it should be configured like that, but now I slightly increased the slider to [this](http://i.imgur.com/3SiLPuV.png) and I think it seems to help–the LEDs don't really seem to flash anymore like they once did when I yawed or rolled not extremely. However, when I rotate my head above a certain degree upward, the top LED usually disappears but I have a feeling it may be the sun visor of my cap that is interfering and not the threshold (I think it used to be like that with the previous settings too).
That said, the sync issue when rotating over 90 degrees to the left or right still seems to be present.

@Len62 I just tried that but I have a feeling the min size is not really the cause of the issue because the LEDs are almost always 2-3 or up and not lower than that (my current min/max is 1.0/50.0, respectively).

Thanks to both of you for helping so far, I really appreciate it.

sthalik, 2017-07-16:
@OCTIAT there's an absurd bug where pose was zeroed once it stopped detecting. Grab today's build from http://opentrack.misaki.pl.

closed 2017-09-07 by sthalik