opentrack/opentrack #654: Installer causes bitdefender to trigger on the file opentrack-compat.dll

cofilippo, 2017-07-15:
Somehow this file is triggering Bitdefender to block installation. It says that it contains Gen:Variant.Razy.196287

sthalik, 2017-07-16:
Which files from the portable archive do this?

cofilippo, 2017-07-26:
Hello Stanislav, 

I'm sorry to respond this late. The file is the same file, opentrack-compat.dll. I actually tried overwriting the existing installation before starting frech. Somehow it triggered in copying this file every time, even after adding an exclusion to the installation folder in BD. I eventually managed to install opentrack by rebooting  (to reset BitDefender, quite annoying but quite often the only wat to stop it from tripping up after an initial warning, even after adding exclusions)
I've also contacted bitdefender support and they say they're looking into it as well. I suspect that overwriting the file triggered BD's heuristics or something. I'll update the issue if I have any news from them.

BTW, I looked through the code when I saw that you've reintroduced the multiple tracking again (very welcome feature for me, I even considered doing a PR myself at some point) I was wondering why you use Euler angles. Euler angles can cause gimbal lock. Maybe I missed something and you actually do use a matrix somewhere. It can help solve gimbal lock when you use a matrix and multiply this matrix with quarternions to do the calculations. (If you happen to think gimbal lock is a problem in the first place, that might not be the case)

Best regards,

Onno Filippo

-------- Origineel bericht --------

https://github.com/opentrack/opentrack/issues/654#issuecomment-315583055

sthalik, 2017-07-26:
With regard to antivirus signatures, it's the entire signature that's too wide. If they only whitelist one version, another version or a dev build will trigger again. That was the problem with Avira that's now hopefully fixed.

We're using Euler angles only because it was like that since the beginning. Otherwise we'd have to go over all trackers, filters, and protocols in detail.

Furthermore, we can extract a sort-of-canonical representation of Euler angles by

- dividing each of the ypr triple by a power of two[1], 4 or more[2]
- converting to matrix form
- converting back
- multiplying ypr by that power of two[1]

Also keep in mind that protocols operate on Euler angles directly. SimConnect, freetrack, mouse, fsuipc, joystick, and I can go on. We'd need to convert to Euler angles anyway and take care of possible gimbal lock.

Gimbal lock is just having many representations for a particular rotation. By that, it's not a show-stopper of any kind. If we were applying accumulating rotation to a body it would be a problem.

We could also arrive at the same solution by converting to axis-angle form, then dividing the angle by a power of two. Finally converting to Euler and multiplying back.

[1] Can be any number but a power of two is more accurate with the binary floating-point representation.
[2] Need to prevent any axis of rotation from getting to 90.

cofilippo, 2017-07-31:
BD got back to me. They say they've fixed it on their end and that I should update my system. They didn't provide any details about what they changed.

closed 2017-07-31 by cofilippo