opentrack/opentrack #655: TrackHat opentrack

rockandrole18, 2017-07-16:
Hey all, I have recently bought a TrackHat head-tracking kit with a hat and a camera from trackhat.org, and I noticed that in addition to the TrackHat opentrack that they recommend to use, there's the source version on here which is updated more regularly and seems to have more options than the TrackHat opentrack. Because I was having issues with setting up the TrackHat opentrack, maybe I should give a try to the source version? What do you think?
I think I'll be downloading it anyway when I get home, but should I configure anything differently than suggested in the quick start guide because I have a TrackHat, or is the procedure is pretty much the same?
Thank you very much in advance 👍

Len62, 2017-07-16:
No harm in giving it a try. If you run into any problems  just ask for some help.

sthalik, 2017-07-16:
@rockandrole18 the trackhat branch has defaults set up that you needn't change. In opentrack you need to input values such as clip/cap size, camera fps, etc.

sthalik, 2017-07-16:
FWIW, here's the default clip measure:

[image]

rockandrole18, 2017-07-16:
@sthalik Any idea what the cap model dimensions are? (I bought a hat)

sthalik, 2017-07-17:
@rockandrole18 see https://github.com/opentrack/opentrack/blob/trackhat/tracker-pt/point_tracker.h#L88

sthalik, 2017-07-17:
And I don't know which is which, sorry.

rockandrole18, 2017-07-17:
@stahlik Thanks, I think I figured it out. I'll let you know if I run into problems but for now, I think I'll close this thread.

closed 2017-07-17 by rockandrole18