opentrack/opentrack #656: trembling problems opentrack

xolights, 2017-07-18:
Hi (excuse my english i'm french),

I have a big problems with open track :

I use windows 10 and a galaxy s8 plus

I have install apk and open track on my pc but i have this trembling effect all time : https://gyazo.com/0dae1317dc93d9fc36796282f8ec611a

You have any idea about my problems ?

Thx

sthalik, 2017-07-18:
The phone has a gyro so it has to work. Try removing headphones, speakers, and other magnets from the room. I'm not saying they can't be in the room, but let's check the hypothesis that it's magnets breaking magnetometer readings.

xolights, 2017-07-18:
Yeah i have remove all magnet all my sensor moving i dont hknow for what ...

sthalik, 2017-07-18:
If you position the device relative to the headphones in a right way, magnetometer readings should be correct enough. Do you have speakers? What other magnets do you have?

Technically we could do AHRS with pure gyro+acc, disregarding magnetometer readings completely. Please try repositioning the device and if it doesn't help, I'll do something with AHRS.

Hint: you have headphones on your ears, so position the device on top of your head if possible,

xolights, 2017-07-18:
yes i have positioning my smartphone on the top of my head but same problem :(

sthalik, 2017-07-18:
Will you be able to test an .apk build this day?

xolights, 2017-07-18:
accelerometer moving all time

xolights, 2017-07-18:
yes i have resr apk

xolights, 2017-07-18:
yes i can test apk

sthalik, 2017-07-18:
What's up with the accelerometer? It should be affected by gravity when held stationary at a vector dependent upon the phone's own orientation.

xolights, 2017-07-18:
i dont move move my phone but acceleromrometre moving all time

sthalik, 2017-07-18:
Gravity moves it :)

sthalik, 2017-07-18:
Now wait a minute, does it move in different directions if you don't move it?

xolights, 2017-07-18:
ok so its normal for accelerometre, but why other sensor moving ?

xolights, 2017-07-18:
yes is moving in different direction

sthalik, 2017-07-18:
Something's very wrong here since with the phone lying on the table, you should get one of 3 accelerometer DOF pointing at -10 and others stationary.

sthalik, 2017-07-18:
Are you able to verify with CPU-Z what acc is doing?

xolights, 2017-07-18:
euh no sorry :(

xolights, 2017-07-18:
all time same problems https://gyazo.com/10b38798f9f419a0ad23af672f7c5f47

sthalik, 2017-07-18:
Does it behave like this without magnets in the room, also?

xolights, 2017-07-18:
yes

sthalik, 2017-07-18:
Also it's strange as hell that it's only the pitch DOF that gets any data. Rest is zeroes.

sthalik, 2017-07-18:
Please install CPU-Z from Play Store and see what the sensors are doing.

xolights, 2017-07-18:
ok

xolights, 2017-07-18:
sensor page : 
[image]
[image]

sthalik, 2017-07-18:
If you rotate the phone continuously (can be slightly), do you get gyroscope readings? These are delta readings, hence the need fo rotate while taking readings.

sthalik, 2017-07-18:
Also look at the "MPL Rotation Vector" readings and see if they're correct when keeping the phone on various orientations.

xolights, 2017-07-18:
yes

xolights, 2017-07-18:
nono gyrocope dont work

xolights, 2017-07-18:
and rotation vector enable

albertolg, 2017-07-19:
tried calibrating the sensors?

sthalik, 2017-07-20:
@xolights I can't get anything to work without at least gyro and acc working.

closed 2017-09-07 by sthalik