opentrack/opentrack #657: Remember the curves in 2.3.8

petsild, 2017-07-23:
Hi,
in the new 2.3.8 does not remember the settings of the curves. After switching off are all in the basic settings. Pet

Len62, 2017-07-23:
I was just about to report the same issue.

MathijsG, 2017-07-23:
Yeah same problem here :(. Maybe we should change something about the silent saving too? Because when this encounters all your mappings are killed already, only when loading the config.

Marduk879, 2017-07-23:
Same issue here. Please fix.

OCTIAT, 2017-07-24:
Yeah, I have it too.. The Accela settings also get reset every time I restart the program.

Len62, 2017-07-24:
There's a new nightly build: https://www.dropbox.com/sh/823dd4wyhixv5a7/AADtsPX4MwEJ_UtmsWJBTraUa/opentrack-test-20170724_01.7z?dl=0

It fixes the issue with the settings.

petsild, 2017-07-24:
This build works correctly, the curves and Accela after disabling saves the settings.
Thank you for the quick fix.  Pet

sthalik, 2017-07-25:
I pulled the build after early complaints about broken sliders and point lists.

Apologies for your lost settings. Every dialog you saved by the OK button lost their sliders. Same as for the mapping window. I fucked up this time.

MathijsG, 2017-07-25:
Hey man, don't be so hard on it. In the meantime we should think about a proper way for you and the community to do some quality assurance before every release I think?

Len62, 2017-07-25:
As Mathisj said, don't be too hard on yourself! Tried the 2.3.9 test build and things look okay.

closed 2017-09-07 by sthalik