opentrack/opentrack #658: Couldn't make new release track correcty with a Floppy disk filter as before.

Forsaken-Player, 2017-07-23:
I've always used opentrack (PS3 cam without IR filter) with a floppy disk filter. It was tracking correctly. Now with this new system I have to remove the floppy disk to make it work. Was this a wanted behavior? Tried a lot of cam settings and there's no way with the light filter.

Len62, 2017-07-23:
Are you sure it's not related to the mappings being set to default, as some of us are experiencing? See the thread about curves. I'm running the same set-up as you and have not run into the problem you are having.  Do the leds display correctly ? Could be the threshold needs adjusting in pointracker.

Forsaken-Player, 2017-07-23:
@Len62 No, it doesn't track the leds correctly with the light filter. Works great without it.

Len62, 2017-07-24:
With the filter on do the led's show at all? Not the normal dots but just a light gray spot. Is it possible for you to maybe capture a short video, with, and without the filter?

MathijsG, 2017-07-24:
I have the same setup but the dots do get recognised here. Although the dots appear much smaller and it also spots a lot of unwanted light sources (while it's not even sunny at all). Looks like something has changed indeed.

Len62, 2017-07-24:
Strange, for sure. Have you checked the cleye settings to see what is visible? Also, maybe a need to re-set the threshold information in point tracker? Just throwing out some ideas, the new nightly build is working fine for me.

Forsaken-Player, 2017-07-24:
Checked every cam settings. With the light film (aka floppy disk filter) under all the previous releases it tracked fine. Now it loses track of some leds. I will try to take a picture later when I get home from work. What I did noticed is that when I use the filter the camera detects 2 different light spots at the led and the software might not be able determine the center. Losing that led track.

Laserclip, 2017-07-24:
I confirm - in Opentrack 2.3.1.5 I have 7.8 px, 8 px 8.7px in 2.3.8   7.4 px 7.5 px 8.3 px - values in the same distance and the same settings are lower

MathijsG, 2017-07-25:
That's very minor to me. What pixel size does @Forsaken-Player  have? It must be a bigger difference than I and @Laserclip  get.

Forsaken-Player, 2017-07-25:
I've made a video but I didn't know where to upload it. I'm using 320x240 resolution, pixels are about 2.5 and 3. But all the time visible and not tracked as before.

Laserclip, 2017-07-25:
Create a gmail account - and you have a virtual drive - upload a video there and share a video link

Len62, 2017-07-26:
I noticed some overlap of two of your leds. Try increasing the resolution to 640x480 and you'll get some more separation. Still puzzling me about the filter issue and led point sizes. How do things look in the cleye driver display?

Forsaken-Player, 2017-07-26:
I use that resolution because of the fps. I don't see any overlap. Here is another fast try, there's no way to use it with the filter even if the points are visible, the software doesn't track them. I'll do a short video with the previous version. 
https://mega.nz/#!9AEyBIyC!OpEBQqmqL5Yn4HJ7XjN5kLscD3qpKnJHgn_QH4t8Gsk

Forsaken-Player, 2017-07-26:
Here is previous version with the filter:
https://mega.nz/#!MF1mxJoD!UfxU4XUKd5F803zHqXyxD2BWa9dDkowdr1CPft5zVS0

Len62, 2017-07-27:
Do you really need 187fps? 640x480 @75 fps works good (not that it is related to the primary issue). I'm still curious as to what the cleye display output looks like. What is the max threshold set at? Almost seems like the led's are under powered.

Forsaken-Player, 2017-07-28:
I always used with that resolution and was great. I like it's response! I think it's a problem in my leds, because the middle one is almost 2 pixels smaller than the other two. 

PS: Might it be possible that the software can't handle such difference? @sthalik

closed 2017-08-01 by Forsaken-Player