opentrack/opentrack #659: [Feature] Make implicit saving optional.

MathijsG, 2017-07-24:
Every once in a while we get a feature breaking bug with a new version. Because it loads the mapping incorrectly (or every other setting for that matter) it's automatically overwriting the bad settings to the mapping, because every change gets saved automatically. Maybe we should make this autosaving optional or maybe even revert it. Now every change is permanent after clicking OK in a window, without an easy way to undo it.

What are people's thoughts on this? Of course I should have backupped it more, but because I thought it was a stable release I forgot about it :-P.

petsild, 2017-07-24:
Maybe I'm doing it wrong but always after setting the Accela filter and curves I prefer to Stop/Start. If it responds OT on all changes of parameters in real mode then the option to return the value of the back had its justification.

MathijsG, 2017-07-24:
But sometimes you are changing more than 1 parameter, or did something accidentally. Usability-wise I find autochange without undo not so user friendly in my workflow. Now hooking up my external drive to restore an earlier backup for my profiles :-P.

Len62, 2017-07-24:
I would like to see this in some form. I like the autosaving for the most part. At what point do we give users the option to save their settings then? Maybe an added drop-down to the Profile to "Save Changes"?

sniperwolfpk5, 2017-07-25:
Well auto save should be the default behavior. As it is standard in desktop based applications.

sthalik, 2017-07-25:
I haven't seen a program letting undo saved used preferences. Except reset to default, which is what ironically happened.

Pressing "cancel" in a dialog not only reverts the settings but avoids saving the .ini altogether.

What we're missing is QA. It's not a program feature that could fix it.

MathijsG, 2017-07-25:
Can't we somehow differentiate between a user altered setting (which we want to autosave) and some wrong readouts from the config itself? Not trying to be rude here, but this is not the first time the mappings were overwritten this way, so when this is a calculated potential issue everytime, can we somehow fix this in the code, by not even loading a config if it gets failed to read?

What could the community contribute with QA? And in what way would that be best? maybe we should have a few weeks before every increment, to officially test out stuff? Instead of just releasing a new version?

And don't be so hard on yourself on the bugs now and then, this software is free, you don't even want donations and you do the coding mostly on your own. :)

sniperwolfpk5, 2017-07-26:
I thing you can QA if you release a beta version and after a week release normal version. Just like I see it in DCS world. They release update on beta branch if every thing is smooth then they release normal version. Just put a disclaimer

sthalik, 2017-09-07:
I'll continue posting pre-release builds on this bug tracker, as earlier.

closed 2017-09-07 by sthalik