opentrack/opentrack #660: [Feature] Support for Dual Gyro and accelerometer tracking.

Hacksawfred3232, 2017-07-25:
Sorry if i spelt anything wrong. The reason i am asking for this is because Trinus VR does not support in the basic terms, "Leaning in to look at stuff more clearly." Which basically means using something like a acceleromter to detect that someone is trying to lean in and look at a object more closely. 
And before anyone mentions this in there comment, i am asking for DUAL tracking of both the Gyro and accelerometer, NOT to prefer tracking of the acceleromter separately. You use the Gyro for the normal turning of the head, and you use the accelerometer for movement of the head that is not Pitch, Yaw or Roll. So basically the leaning in bit. Almost every phone on the market today has a Gyro and a accelerometer. I am pretty sure you can do this.

sthalik, 2017-07-25:
You can't do this with inertial sensors alone. The error will accumulate so fast that it's better not to track position.

Also there's the "fusion" tracker. It's enough you use it for purposes.

Hacksawfred3232, 2017-07-25:
I see. Okay then. I will investigate the "fusion" tracker method. Thanks for your help!

closed 2017-07-25 by Hacksawfred3232