opentrack/opentrack #661: Compilation on linux does not work

ljgdasfhk, 2017-07-27:
arch, gcc 7.1

> 
[  0%] Building CXX object CMakeFiles/opentrack-version.dir/version.cpp.o
[  1%] Linking CXX static library libopentrack-version.a
[  1%] Built target opentrack-version
make[2]: *** No rule to make target `/home/someone/opentrack-opentrack-2.3.8/tracker-udp/lang/nl_NL.ts', needed by `nl_NL.qm'.  Stop.
make[1]: *** [CMakeFiles/i18n.dir/all] Error 2
make: *** [all] Error 2
>

ljgdasfhk, 2017-07-27:
`[ 20%] Building CXX object options/CMakeFiles/opentrack-options.dir/bundle.cpp.o
In file included from /home/sobolev/opentrack-opentrack-2.3.8/options/group.hpp:5:0,
                 from /home/sobolev/opentrack-opentrack-2.3.8/options/bundle.hpp:11,
                 from /home/sobolev/opentrack-opentrack-2.3.8/options/bundle.cpp:9:
/home/sobolev/opentrack-opentrack-2.3.8/compat/util.hpp:121:41: error: ‘always_inline’ was not declared in this scope
 #   define always_inline __attribute__((always_inline, gnu_inline))
                                         ^
/home/sobolev/opentrack-opentrack-2.3.8/compat/util.hpp:121:56: error: ‘gnu_inline’ was not declared in this scope
 #   define always_inline __attribute__((always_inline, gnu_inline))
                                                        ^
/home/sobolev/opentrack-opentrack-2.3.8/compat/util.hpp:121:56: note: suggested alternative: ‘getline’
make[2]: *** [options/CMakeFiles/opentrack-options.dir/build.make:135: options/CMakeFiles/opentrack-options.dir/bundle.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:807: options/CMakeFiles/opentrack-options.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
`

ljgdasfhk, 2017-07-27:
SOME WARNINGS

`[ 64%] Building CXX object gui/CMakeFiles/opentrack-user-interface.dir/mapping-window.cpp.o                                                                             
In file included from /home/sobolev/opentrack-opentrack-2.3.8/options/options.hpp:18:0,
                 from /home/sobolev/opentrack-opentrack-2.3.8/logic/mappings.hpp:11,
                 from /home/sobolev/opentrack-opentrack-2.3.8/gui/mapping-window.hpp:3,
                 from /home/sobolev/opentrack-opentrack-2.3.8/gui/mapping-window.cpp:9:
/home/sobolev/opentrack-opentrack-2.3.8/options/tie.hpp: In instantiation of ‘struct options::tie_setting(options::value<t>&, QComboBox*)::<lambda(int)> [with t = axis_opts::max_clamp]::<lambda()>’:
/home/sobolev/opentrack-opentrack-2.3.8/options/tie.hpp:39:47:   required from ‘options::tie_setting(options::value<t>&, QComboBox*)::<lambda(int)> [with t = axis_opts::max_clamp]’
/home/sobolev/opentrack-opentrack-2.3.8/options/tie.hpp:37:30:   required from ‘struct options::tie_setting(options::value<t>&, QComboBox*) [with t = axis_opts::max_clamp; typename std::enable_if<std::is_enum<_Tp>::value>::type = void]::<lambda(int)>’
/home/sobolev/opentrack-opentrack-2.3.8/options/tie.hpp:34:24:   required from ‘typename std::enable_if<std::is_enum<_Tp>::value>::type options::tie_setting(options::value<t>&, QComboBox*) [with t = axis_opts::max_clamp; typename std::enable_if<std::is_enum<_Tp>::value>::type = void]’
/home/sobolev/opentrack-opentrack-2.3.8/gui/mapping-window.cpp:29:51:   required from here
/home/sobolev/opentrack-opentrack-2.3.8/options/tie.hpp:40:50: warning: ‘options::tie_setting(options::value<t>&, QComboBox*)::<lambda(int)> [with t = axis_opts::max_clamp]::<lambda()>’ declared with greater visibility than the type of its field ‘options::tie_setting(options::value<t>&, QComboBox*)::<lambda(int)> [with t = axis_opts::max_clamp]::<lambda()>::<v capture>’ [-Wattributes]
                                                [&]() {
                                                  ^`

`[ 83%] Building CXX object tracker-hatire/CMakeFiles/opentrack-tracker-hatire.dir/ftnoir_tracker_hat_dialog.cpp.o                                                       
In file included from /home/sobolev/opentrack-opentrack-2.3.8/options/options.hpp:18:0,
                 from /home/sobolev/opentrack-opentrack-2.3.8/tracker-hatire/ftnoir_tracker_hat_settings.h:11,
                 from /home/sobolev/opentrack-opentrack-2.3.8/tracker-hatire/thread.hpp:4,
                 from /home/sobolev/opentrack-opentrack-2.3.8/tracker-hatire/ftnoir_tracker_hat.h:3,
                 from /home/sobolev/opentrack-opentrack-2.3.8/tracker-hatire/ftnoir_tracker_hat_dialog.cpp:11:
/home/sobolev/opentrack-opentrack-2.3.8/options/tie.hpp: In instantiation of ‘struct options::tie_setting(options::value<t>&, QComboBox*)::<lambda(int)> [with t = QSerialPort::BaudRate]::<lambda()>’:
/home/sobolev/opentrack-opentrack-2.3.8/options/tie.hpp:39:47:   required from ‘options::tie_setting(options::value<t>&, QComboBox*)::<lambda(int)> [with t = QSerialPort::BaudRate]’
/home/sobolev/opentrack-opentrack-2.3.8/options/tie.hpp:37:30:   required from ‘struct options::tie_setting(options::value<t>&, QComboBox*) [with t = QSerialPort::BaudRate; typename std::enable_if<std::is_enum<_Tp>::value>::type = void]::<lambda(int)>’
/home/sobolev/opentrack-opentrack-2.3.8/options/tie.hpp:34:24:   required from ‘typename std::enable_if<std::is_enum<_Tp>::value>::type options::tie_setting(options::value<t>&, QComboBox*) [with t = QSerialPort::BaudRate; typename std::enable_if<std::is_enum<_Tp>::value>::type = void]’
/home/sobolev/opentrack-opentrack-2.3.8/tracker-hatire/ftnoir_tracker_hat_dialog.cpp:95:52:   required from here
/home/sobolev/opentrack-opentrack-2.3.8/options/tie.hpp:40:50: warning: ‘options::tie_setting(options::value<t>&, QComboBox*)::<lambda(int)> [with t = QSerialPort::BaudRate]::<lambda()>’ declared with greater visibility than the type of its field ‘options::tie_setting(options::value<t>&, QComboBox*)::<lambda(int)> [with t = QSerialPort::BaudRate]::<lambda()>::<v capture>’ [-Wattributes]
                                                [&]() {
                                                  ^
/home/sobolev/opentrack-opentrack-2.3.8/options/tie.hpp: In instantiation of ‘struct options::tie_setting(options::value<t>&, QComboBox*)::<lambda(int)> [with t = QSerialPort::DataBits]::<lambda()>’:
/home/sobolev/opentrack-opentrack-2.3.8/options/tie.hpp:39:47:   required from ‘options::tie_setting(options::value<t>&, QComboBox*)::<lambda(int)> [with t = QSerialPort::DataBits]’
/home/sobolev/opentrack-opentrack-2.3.8/options/tie.hpp:37:30:   required from ‘struct options::tie_setting(options::value<t>&, QComboBox*) [with t = QSerialPort::DataBits; typename std::enable_if<std::is_enum<_Tp>::value>::type = void]::<lambda(int)>’
/home/sobolev/opentrack-opentrack-2.3.8/options/tie.hpp:34:24:   required from ‘typename std::enable_if<std::is_enum<_Tp>::value>::type options::tie_setting(options::value<t>&, QComboBox*) [with t = QSerialPort::DataBits; typename std::enable_if<std::is_enum<_Tp>::value>::type = void]’
/home/sobolev/opentrack-opentrack-2.3.8/tracker-hatire/ftnoir_tracker_hat_dialog.cpp:96:52:   required from here
/home/sobolev/opentrack-opentrack-2.3.8/options/tie.hpp:40:50: warning: ‘options::tie_setting(options::value<t>&, QComboBox*)::<lambda(int)> [with t = QSerialPort::DataBits]::<lambda()>’ declared with greater visibility [... cut ...]
/home/sobolev/opentrack-opentrack-2.3.8/tracker-hatire/ftnoir_tracker_hat_dialog.cpp:99:52:   required from here
/home/sobolev/opentrack-opentrack-2.3.8/options/tie.hpp:40:50: warning: ‘options::tie_setting(options::value<t>&, QComboBox*)::<lambda(int)> [with t = QSerialPort::StopBits]::<lambda()>’ declared with greater visibility than the type of its field ‘options::tie_setting(options::value<t>&, QComboBox*)::<lambda(int)> [with t = QSerialPort::StopBits]::<lambda()>::<v capture>’ [-Wattributes]
`

sthalik, 2017-07-28:
> make[2]: *** No rule to make target /home/someone/opentrack-opentrack-2.3.8/tracker-udp/lang/nl_NL.ts', needed bynl_NL.qm'. Stop.

The file exists. Incomplete checkout?

> /home/sobolev/opentrack-opentrack-2.3.8/compat/util.hpp:121:41: error: ‘always_inline’ was not declared in this scope

It's been fixed on July 24.

The warnings are irrelevant. I should add a pragma.

ljgdasfhk, 2017-07-28:
I downloaded the archive 'Source code (tar.gz)'

sthalik, 2017-07-28:
Don't. It's a useless thing github adds no matter what. Clone the "unstable" branch.

ljgdasfhk, 2017-07-28:
ok

ljgdasfhk, 2017-07-28:
git clone ....

Scanning dependencies of target opentrack-version
[  0%] Building CXX object CMakeFiles/opentrack-version.dir/version.cpp.o
[  1%] Linking CXX static library libopentrack-version.a
[  1%] Built target opentrack-version
Scanning dependencies of target i18n
make[2]: *** Нет правила для сборки цели «/home/sobolev/opentrack/proto-libevdev/lang/nl_NL.ts», требуемой для «nl_NL.qm».  Останов.
make[1]: *** [CMakeFiles/Makefile2:233: CMakeFiles/i18n.dir/all] Ошибка 2
make: *** [Makefile:130: all] Ошибка 2

ljgdasfhk, 2017-07-28:
git clone https://github.com/opentrack/opentrack

Scanning dependencies of target opentrack-version
[  0%] Building CXX object CMakeFiles/opentrack-version.dir/version.cpp.o
[  1%] Linking CXX static library libopentrack-version.a
[  1%] Built target opentrack-version
Scanning dependencies of target i18n
make[2]: *** No rule to make target '/home/sobolev/opentrack/proto-libevdev/lang/nl_NL.ts', needed by 'nl_NL.qm'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:233: CMakeFiles/i18n.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

ljgdasfhk, 2017-07-28:
git clone -b unstable --recursive https://github.com/opentrack/opentrack

[sobolev@computer ot]$ make
Scanning dependencies of target opentrack-version
[  0%] Building CXX object CMakeFiles/opentrack-version.dir/version.cpp.o
[  1%] Linking CXX static library libopentrack-version.a
[  1%] Built target opentrack-version
Scanning dependencies of target i18n
[  1%] Running lrelease for nl_NL
[  1%] Running lrelease for ru_RU
[  1%] Running lrelease for stub
[  1%] Built target i18n
[  1%] Generating moc_variance.cpp
[  2%] Generating moc_shm.cpp
[  2%] Generating moc_camera-names.cpp
[  2%] Generating moc_check-visible.cpp
[  2%] Generating moc_euler.cpp
[  3%] Generating moc_export.cpp
[  3%] Generating moc_functional.cpp
[  3%] Generating moc_linkage-macros.cpp
[  3%] Generating moc_linkage.cpp
[  4%] Generating moc_math-imports.cpp
[  4%] Generating moc_meta.cpp
[  4%] Generating moc_nan.cpp
[  4%] Generating moc_ndebug-guard.cpp
[  6%] Generating moc_powerset.cpp
[  6%] Generating moc_process-list.cpp
[  6%] Generating moc_qcopyable-mutex.cpp
[  6%] Generating moc_run-in-thread.cpp
[  7%] Generating moc_simple-mat.cpp
[  7%] Generating moc_sleep.cpp
[  7%] Generating moc_time.cpp
[  7%] Generating moc_timer.cpp
[  8%] Generating moc_util.cpp
Scanning dependencies of target opentrack-compat
[  8%] Building CXX object compat/CMakeFiles/opentrack-compat.dir/camera-names.cpp.o
[  8%] Building CXX object compat/CMakeFiles/opentrack-compat.dir/check-visible.cpp.o
/home/sobolev/opentrack/compat/check-visible.cpp:56:1: ошибка: «always_inline» does not name a type; did you mean «__always_inline»?
 always_inline OTR_COMPAT_EXPORT
 ^~~~~~~~~~~~~
 __always_inline
/home/sobolev/opentrack/compat/check-visible.cpp:61:1: ошибка: «always_inline» does not name a type; did you mean «__always_inline»?
 always_inline OTR_COMPAT_EXPORT
 ^~~~~~~~~~~~~
 __always_inline
make[2]: *** [compat/CMakeFiles/opentrack-compat.dir/build.make:175: compat/CMakeFiles/opentrack-compat.dir/check-visible.cpp.o] Ошибка 1                               
make[1]: *** [CMakeFiles/Makefile2:446: compat/CMakeFiles/opentrack-compat.dir/all] Ошибка 2                                                                            
make: *** [Makefile:130: all] Ошибка 2

ljgdasfhk, 2017-07-28:
1. After the work, the translation files are deleted.
2. In the folder proto-libevdev there are no translation files.

sthalik, 2017-07-29:
The translation files get created and regenerate with the Ninja generator. But the `make(1)` generator requires the files to exist at all beforehand, apparently.

sthalik, 2017-07-29:
It should build fine. I did a test build with all working modules and nothing bad seemed to happen.

sthalik, 2017-07-29:
Please quote the error text rather than pasting it inline in messages. It breaks due to Markdown formatting.

closed 2017-07-29 by ljgdasfhk

ljgdasfhk, 2017-07-29:
Now we have to start with LD_LIBRAR_PATH = / ...
Otherwise it does not find its libraries .so
In previous versions it was not

sthalik, 2017-07-30:
It should work after `make install'.