opentrack/opentrack #662: Hatire Arduino plugin don't read data from serial. Bug?

r57zone, 2017-07-28:
I flashed firmware [Razor AHRS](https://github.com/Razor-AHRS/razor-9dof-ahrs/tree/master/Arduino/Razor_AHRS) to Arduino + GY 85. I [changed the speed of 115200](http://i.imgur.com/QKRJmYL.jpg) in the plugin and sketch Arduino. I changed the com-port in the Hatire Arduino plugin. In OpenTrack [don't read data](http://i.imgur.com/B0mdaQh.jpg). Arduino sends [data to the com port "YPR ="](http://i.imgur.com/8wLGUh3.jpg). In the sketch, the Razor_AHRS_test for Processing [also works](http://i.imgur.com/Iqduumo.jpg).

From log file "HATDiagnostics.txt":
> 13:38:56.179: Starting Tracker
> 13:38:56.195: Setting serial port name
> 13:38:56.196: Opening serial port
> 13:38:56.206: Port Open
> 13:38:56.219: Port Parameters set
> 13:38:56.219: Raising DTR
> 13:38:56.224: Raising RTS
> 13:38:56.224: Couldn't set RTS
> 13:38:56.224: Waiting on init
> 13:38:56.224: Port setup, waiting for HAT frames to process

r57zone, 2017-08-02:
It seems the latest firmware is not compatible with OpenTrack. I'll try the old firmware.

MrCapone, 2017-08-06:
I use same tracker based on Arduino nano + GY85. Do you change output format to face track on Razor_AHRS?
Also try com port monitor on Arduino IDE, you should see something like that
[image]

r57zone, 2017-08-06:
@MrCapone This is Binary output, yes I changed arduino sketch, but it does not work.  Could you upload your firmware somewhere?

MrCapone, 2017-08-07:
@r57zone [link](https://mega.nz/#!oE4gzC5A!Ssnny6eX3Yb7y-CyjX4tKynobVMbXs9uRkRtlvwXrks)
On `Razor_AHRS.ino:192` change `int output_format` to `OUTPUT__FORMAT_FACETRACK`

r57zone, 2017-08-07:
@MrCapone I replaced, I flashed you sketch, changed the speed, checked Com-port, but OpenTrack still does not see.

In addition, the latest version of Razor IMU does not support FACETRACK. Perhaps it is worth changing the Hatire Arduino plugin to read the text #ypr = ...

MrCapone, 2017-08-07:
@r57zone Just tested same sketch from my link and everything works.
[image]
Did you change `#define OUTPUT__BAUD_RATE` to 115200 on line 170?
Also you can try [this sketch](https://mega.nz/#!QAQFBYLY!8jLGcPLapgEF9XtkgwT-dA8-7jP3LmJfUjpKe2sB4Wg). It was made just for Hatire plugin.

r57zone, 2017-08-07:
@MrCapone yes. i changed speed in sketch and in OpenTrack

MrCapone, 2017-08-07:
@r57zone Dont know what your problem. But this is definitely not opentrack's bug or issue. Maybe some other soft use this com port at same time?

r57zone, 2017-08-07:
No, the port is not busy.
[image]
[image]
[image]
[image]
[image]
[image]

MrCapone, 2017-08-07:
Arduino IDE can use com port. Did you close it before use opentrack? I also tried stable opentrack release on windows: 
[image]

r57zone, 2017-08-07:
Yes, I closed Arduino Ide. I've checked many times.

MrCapone, 2017-08-07:
I give up. I really dont know what your problem is, but id definitely not opentrack's bug. I can only give you my [opentrack build](https://mega.nz/#!Qdhhib7D!uCd6tX02xIymSKSo8Yp2MAe44EBdp8-37GCK2Zouyq8) and my [opentrack settings](https://mega.nz/#!UBgjgRRK!OEL3Yc-ZvNygSRprbs5v4GmUBoHCApZ4kW3wrKl0jf4) (place it on `Documents\opentrack-2.3`). Maybe it will help.

sthalik, 2017-08-08:
Just to rule out the obvious, did you compile and upload the compiled firmware?

r57zone, 2017-08-08:
@sthalik yes, after the changes, I saved, compiled and uploaded to arduino.

sthalik, 2017-08-08:
Are you guys using the same GPIO port, sensor types, and wiring?

r57zone, 2017-08-08:
I used GY-85 BMP085 (ITG3205 + ADXL345 + HMC5883L) + Arduino Nano V3 (CH340). 
Maybe can add output text support to the plugin? That someone has a [similar problem](https://github.com/opentrack/opentrack/issues/626), if there was a text, then the problem would not arise.

sthalik, 2017-08-08:
Or use Python or minicom.

closed 2017-08-08 by r57zone

sthalik, 2017-08-08:
@MrCapone can you comment on the hardware?

sthalik, 2017-08-08:
Also as I saw the first firmware releases -- sadly they're badly written. It should be easy to make sending via wire, reading from sensors, pose estimation all modular. As I saw the original firmware, there were tons of static arrays but no use of the `struct` keyword.

I can help but with no ability to run the code, other than compile-testing.

MrCapone, 2017-08-08:
@sthalik I use exactly same hardware. Arduino nano V3 based on Atmega 328 with CH340 USB chip: 
[image]
and GY-85 
[image]
GPIO (GY-85 -> Arduino)
VCC_IN -> +5V
GND -> GND
SCL -> A5
SDA ->A4
G_INT -> D2

r57zone, 2017-08-28:
If someone repeats a similar problem, then you can use the [vr proxy apps](https://github.com/r57zone/VR-tracking-apps) for SteamVR and GoPro VR Player.

ki714, 2017-09-28:
Did you folks managed to get it working? If not, trying this solutions. It worked for me. 
https://sourceforge.net/p/facetracknoir/discussion/iht/thread/9cbb2f4c/

fzxj520, 2018-02-19:
Try the following items:
https://sites.google.com/site/diyheadtracking/home/diy_ahrs_razor_20170701
I did not test!

sthalik, 2018-02-21:
Does it work right now? @huliqan said it doesn't since 2.3.1.

ki714, 2018-02-21:
I did managed to to get it working with the latest version of opentrack/facetracknoir with Pololu MinIMU V9, See my previous post  Sept 28 2017

huliqan, 2018-02-21:
I testing

sthalik, 2018-02-24:
@huliqan,

It should all work in the recent build (see 2.3.10 prerelease on http://opentrack.misaki.pl). There should be no need to use an old build.

Let's work on fixing the problem so you (and possibly others) don't need an old build.

huliqan, 2018-02-24:
Ok! I Testing
Ok! I Testing

huliqan, 2018-02-24:
hatire and Oculus runtime 1.4.0 works fine!

sthalik, 2018-02-25:
@huliqan this is great news. Also the CPU usage workaround is now automatic.

huliqan, 2018-02-25:
CPU usage 1-2%

dieghito, 2022-01-13:
Salutations,
i also have problems with hatire plugin for arduino ... it does not read what is sent by arduino itself.
Maybe in the various updates the reading protocol has been changed ... may I ask how the arduino data should be sent?
In case I modify my Arduino sketch, so I make it compatible.

Thanks for any kind of support you can give me!

ki714, 2022-01-21:
Did you read this post? I uploaded it in the Cliff of Dover forum as well.
Let me know if you need more help. It's been a while.
Regards
Ki714

ki714, 2022-01-21:
https://theairtacticalassaultgroup.com/forum/showthread.php?t=29072

ki714, 2022-01-21:
https://theairtacticalassaultgroup.com/forum/showthread.php?t=28730