opentrack/opentrack #664: opentrack 2.3.9 release process

sthalik, 2017-07-29:
Hey, It's been a while.

I had to pull 2.3.8 due to sliders and mappings resetting themselves in each .ini file.

This build is meant to become 2.3.9. It's only been lightly tested. Please report any issues, in particular with `tracker/pt` following the change to contours.

- - -

Here's a provisional changelog following from 2.3.8:

- **fix sliders and mappings resetting in configs** -- reported as #657 and on IL-2 Sturmovik forum; apologies to all affected!
- tracker/pt: use contours rather than flood fill to reduce CPU usage -- now with zeroing out non-contour pixels
- tracker/pt: better guard against false detections -- behavior from 2.3.7
- tracker/pt: reduce microstutter
- tracker/pt: change min and max point size -- behavior close to 2.3.7
- tracker/pt: scale min and max point size with resolution
- don't update main window if it's minimized or hidden by another window -- together with tracker/pt changes, CPU usage is now 20% of one core rather than 32% with 60 Hz on my 4-core hyperthreading system

- fix llvm build on mingw-w64
- fix build on Linux -- issue #661, also some other changes

- - - 

**Download**

Grab the prerelease at <<http://opentrack.misaki.pl>>. Starting with today, July 29, the builds are marked as 2.3.9 prereleases.

cheers,
sh (how did we get into this mess?)

albertolg, 2017-07-29:
@sthalik I tried latest build opentrack-test-20170729-prerelease2 and something is wrong I think Changing profile don't change the Accela values.. this with the 2.3.7 profiles... but only if change some value then click ok when I reload the profile the new values are there...

this is what I see in my 2.3.7
[image]

and this is what I see with the opentrack-test-20170729-prerelease2
[image]

same profile

Len62, 2017-07-29:
Seeing the same here.

sthalik, 2017-07-29:
@albertolg this isn't the case anymoree in pre3. For the same reason, `2.3.8+ -> pre2` configs don't have the right sliders anymore.

Len62, 2017-07-29:
Pre3 accela values loading correctly with this build.

petsild, 2017-07-30:
Sometimes happens that the Z axis does not respond while the other directions are working.
I use often the function Alter the centered position sent this games.
I'll fix it either shortcut Center or I have to zoom out and then the Z axis works again.
Maybe it will go some way to eliminate the thank you for the response.

sthalik, 2017-07-31:
@petsild see with dynamic pose and neck off. Otherwise Z should respond on `x` of the mapping. Is it still broken then?

- - -

Please report any bugs, random crashes, etc. I'm givin git another few days.

petsild, 2017-07-31:
I found that the size of the points graphically does not respond to distance. The difference point between a maximum and a minimum is only 0.14 pixels.

Len62, 2017-07-31:
On the Z axis the point size, for me, only changes around 1 or less, but movement on the Z axis works fine for me. Isn't the point extraction based on the brightness of the LED's?

sthalik, 2017-07-31:
It only graphically responds to distance if the selected size is too big. It responds by breaking completely.

Forsaken-Player, 2017-07-31:
@sthalik Hi, this new test build totally fixed the issue https://github.com/opentrack/opentrack/issues/658 
I'll try to test a bit more later! 
Thank you!!!!

petsild, 2017-07-31:
Dynamic pose values of 250 – 1000 no effect. Forgot to add that I actually do not respond or Y,X axis.

sthalik, 2017-08-01:
@petsild what doesn't respond on X, Y axis?

petsild, 2017-08-01:
Previous build 2.3.7 behaves the same sometimes stops responding X,Y,Z axis.
The image actually stays hanging at the starting position, but the Pitch, Yaw and roll work normally. 
Work if the camera move to the next level 50mm from the starting position.

petsild, 2017-08-01:
I did a different configuration OT: 
Set invert Z-axis and the same thing had to do in ROF. Now it manifests itself differently and for the better. Now there's a little deadzone that I have to overcome but naturally zoom. 
This does not restrict, just let it be as it is.
I'm so sorry if I slow you down.

sthalik, 2017-08-01:
The image will stop being updated for up to 5 seconds after opentrack was occluded by another window. It takes close to 8% of a single core to display it needlessly, and should help users with dual core CPUs.

It'll start showing itself immediately if you switch focus to the opentrack app. It'll stay for up to 5 seconds if you have the window visible but something else has focus.

What's the deadzone you're mentioning?

petsild, 2017-08-01:
(What's the deadzone you're mentioning?)

I'm still writing about the freezing point of the z axis.

sthalik, 2017-08-02:
Are you talking about translation freezing unless you also rotate?

sthalik, 2017-08-02:
[that comment was incorrect]

sthalik, 2017-08-02:
@petsild is it that your points only shrink or enlarge, but their centers don't move per se? I think that's a bug.

petsild, 2017-08-02:
Minor graphics bug.
[image]

alganthe, 2017-08-03:
Tested with:
opentrack-test-20170729-pre3
Arma 3 1.72.142223 main branch
DCS 1.5.7.7741

using:
PS3 eye with IR filter removed
Trackhat clip plus

Settings:
PointTracker 1.1
freetrack 2.0 Enhanced

Arma 3: Doesn't detect either freetrack or TrackIR, opentrack version 2.3.1.4 works fine
DCS: No issues

sthalik, 2017-08-03:
@alganthe make sure you start tracking before starting the game, when running from given install directory for the 1st time. Otherwise opentrack works with Arma 3 for me. I have the 64-bit version installed from Steam.

sthalik, 2017-08-03:
@petsild thanks, fixed.

sthalik, 2017-08-14:
Initial release done. Thank you for your participation. Fingers crossed that it's not broken anymore.

Len62, 2017-08-14:
Working fine here. Noticed that @petsild graphics bug still present.

sthalik, 2017-08-15:
Damn I see it too. Will look into it.

petsild, 2017-08-15:
My optopus works flawlessly.

sthalik, 2017-08-15:
I'll work on it. For now, as long as it doesn't crash...

sthalik, 2017-08-15:
Remove the `.pdf` extension. It's a dll.

Does replacing this dll help?

[opentrack-pose-widget.dll.pdf](https://github.com/opentrack/opentrack/files/1224300/opentrack-pose-widget.dll.pdf)

petsild, 2017-08-15:
Pink down worse, green :) improved.

[image]
[image]

Len62, 2017-08-15:
The new dll cleaned it up for me.

closed 2018-01-04 by sthalik