opentrack/opentrack #665: Latest Accela 2.3.9

petsild, 2017-07-29:
Hi,
I can't calm the tremor any value of the Accela filter in the last 2.3.9. Change the parameters of the deadzone yaw,pitch,X,Y,Z doesn't help against the original 2.3.7. there is a look of calm.
Thank you for solving.  Pet

sthalik, 2017-07-29:
Temporarily, does increasing rotation nonlinearity help? It works better than the older version of itself.

petsild, 2017-07-29:
The higher the parameter of nonlinearity causes a huge stuttering.

sthalik, 2017-07-29:
@petsild see if today's second build helps any. Looks like contours weren't as precise as flood fill.

Len62, 2017-07-29:
Ran today's prerelease 2 and did not encounter any problems.

sthalik, 2017-07-29:
@Len62 it's fixed in pre3. Sadly the config you made will have sliders reset to default values again.

petsild, 2017-07-29:
Now it behaves gracefully even at low values of the deadzone.

sthalik, 2017-07-30:
Great! Drop by #664 if you need anything else.

closed 2017-07-30 by sthalik