opentrack/opentrack #666: opentrack not receiving data from android phone.

EbonRogue, 2017-08-01:
I am trying to send tracker data to my PC through the Free PIE IMU app on my LG G5 phone.. Theres no data being displayed on the opentrack program.

I have the correct ports and IP Adresses(i know how to atleast ipconfig) and the network doesn't seem to be the problem. I can stream various things between the devices over my wlan like moonlight or image, video, data transfers.

I've tried all version of opentrack from 2.3 and up.

I've also tried various other apps on android such as Wireless IMU and HeadTracker.

Downloaded android app here: 
https://github.com/Xtreamer/FreePie-DiyHt/blob/master/Lib/Android/com.freepie.android.imu.apk

Followed this guide: 
https://github.com/opentrack/opentrack/wiki/Smartphone-Headtracking

I've been digging for a long while to find a solution, but I can't seem to find anything past some coding jargon that doesn't solve my issue.

sthalik, 2017-08-02:
Disable your firewall for testing. If it suddenly works, it lacks a firewall exception.

sthalik, 2017-08-02:
Also see if you can ping your phone and your desktop from the other device.

EbonRogue, 2017-08-02:
Disabling the firewall hasn't changed anything. I've also confirmed that I can ping any devices or vice versa on my network. Don't seem to be able to ping through a port(5555), but I've port forwarded it anyway.
Still no luck.

sthalik, 2017-08-14:
The default port in opentrack is `4242`.

You can't get a response over UDP generally by 'pinging' it.

Deimos2k17, 2017-08-23:
Hi, same problem here, I have two android devices, and basically what changes is the mac address and the local ip, on one it works, on the other it doesn't, my problem is that I want to use the one that's not working, but port forwarding is on and on the right address, firewall is off, I just can't figure out. what else can be.

sthalik, 2017-08-24:
- Ensure wifi connected
- Ensure your PC IP address and Android IP address are in the same subnet
- Check your Android IP address by left-clicking the network name
- Ensure you can ping it and get a response
- You can't receive from 2 phones at once for it to make sense. Saying this just in case.

- Select "send raw" and "debug" to see sensor data at the Android program's bottom. You must select "send raw" for it due to a bug.

It it's still broken despite that, send your Android OS version and model (manufacturer modifies the firmware invasively, typically).

EbonRogue, 2017-08-27:
-Wifi is always on
-No subnetting, just a home network with standard DHCP(though my PC is static, for testing this stuff).
-All IP adresses for any device on the network is 192.168.1.X
-Receiving pings both ways.
-Just one phone transmitting data.
-I have both of those checked. My phone shows that I am sending accelerometer, gyro, magnetometer, and orientation data.

My phone is an LG G5, running Android 7.0. I hope it helps, but I am assuming something weird is up with the networking. Some button, somewhere would probably fix this, because it seems like a(mostly) isolated issue. 

If you can't find out anything, you can close the issue.

Deimos2k17, 2017-08-27:
Here are my models and Android OS:
Works on: LG K500n - Hardware rev. 1.0 - Android 6.0.1 - Kernel 3.10.49 - SW ver. V10e-222-10
Doesn't work on: HUAWEI TAG-L01 - Hardware ver. A - Android 5.1 -Kernel 3.10.65 
While I was trying right now one more time, I discovered that the Huawei model is NOT sending any data from Gyro, Magnetometer or Orientation, only from Accelerometer, while the LG model only misses the Gyro data (it sends out all the rest). I guess this could be part of the main issue at this point.

sthalik, 2017-08-27:
Gyro is necessary for proper tracking. The rudimentary pose data sent with mag+acc only is a developer convenience.

closed 2017-08-27 by sthalik

germanmontedoro, 2022-05-15:
might be the router is blocking the connection, you can try to connect phone and pc with direct wifi connection.
I was in the same problem i installed the "head tracker for opentrack" by tovinst from playstore and worked