opentrack/opentrack #667: Misstakes

iDevPro, 2017-08-03:
1) (opentrack-compat/timer.cpp) Why you are use this overload in .cpp file compilation got error in there?
```
void Timer::otr_clock_gettime(timespec* ts)
**double Timer::otr_clock_gettime(timespec* ts)**
```
2) (opentrack-qt-mini/qxtglobalshortcut_mac.cpp) In function 
```
quint32 QxtGlobalShortcutPrivate::nativeKeycode(Qt::Key keys)
{
    UTF16Char ch;
    // Constants found in NSEvent.h from AppKit.framework
    switch (key) // would be "keys", because you pass keys as parameter, or rename parameter to key
    {
    case Qt::Key_Return:
        return kVK_Return;
    case Qt::Key_Enter:
...
```
Who can support OS X branch of code ?)

iDevPro, 2017-08-03:
Also in <cmath> abs want double, long double or float but not int:
opentrack-spline/spline.cpp
`x = std::abs(x); // x would be converted to double or long double ->  x = std::abs(double(x));`

iDevPro, 2017-08-03:
Also fix isNan isInf on mac:
append to nan.cpp
```
#elif defined(__APPLE__) && defined(__MACH__)
#   include <cmath>
#   define my_isnan std::isnan
#   define my_isinf std::isinf
extern "C" OTR_COMPAT_EXPORT bool __attribute__ ((noinline)) nanp(double x)
```

iDevPro, 2017-08-03:
But after build opentrack, install to folder, and "build from scratch" app. I can't use Face tracking and can't use plugin for x-plane :)

sthalik, 2017-08-04:
in re https://github.com/opentrack/opentrack/issues/667#issuecomment-320059066
```
Also function "abs" want double, long double or float but not int:
opentrack-spline/spline.cpp
x = std::abs(x); // x would be converted to double or long double -> x = std::abs(double(x));
```

No, see <<http://en.cppreference.com/w/cpp/numeric/math/abs>>.

in re https://github.com/opentrack/opentrack/issues/667#issue-247794756

> Why you are use this overload in .cpp file compilation got error in there?

No one runs OSX daily. It's supported worse than the Linux-specific code. Running it in a VM is a bitch too, even Mavericks.

We'd need the following at least to support as well as the Windows version:

- joystick input via QtGamepad. It's way easier than what Windows needs. Still, it hasn't been written.
- OSX-specific keyboard shortcut handling. There are already Windows and X11-specific versions but not for OSX. The present OSX-specific code works only for very few key names.
- someone taking care of it so build errors don't stay up for 3+ months.

sh

sthalik, 2017-08-05:
You're right about `std::abs` after all, thanks.

iDevPro, 2017-08-05:
Ok, but what about face tracking ?
Also I think need virtual joy emulation :(

sthalik, 2017-08-06:
Face tracking isn't implemented for any platform at the moment.

Virtual joystick needs a kernel module that isn't signed. Not only that, the one I found allowed for sending arbitrary HID messages from userspace.

MrCapone, 2017-08-06:
I think this issue can be closed after https://github.com/opentrack/opentrack/pull/668

iDevPro, 2017-08-10:
Re: but facetracking work in linuxtrack project under OSX :)

sthalik, 2017-08-10:
I already have plans to write it with CLandmark. No time.

sthalik, 2017-08-10:
Closing as per @MrCapone.

closed 2017-08-10 by sthalik