opentrack/opentrack #670: The dependence of the sensitivity

petsild, 2017-08-06:
Hi,
motion control has the current OT totally awesome. Long I'm thinking of one function that would control the accuracy even better. If the Z axis could limit the sensitivity of the other axes Pitch, Yaw and Roll.
An example of pushing buttons in the cockpit in DCS, there is a need for very precise movement and the slightest tremor will not allow to activate the required button.
Or zoom on the Z axis is at 100%, very responsive and keep the view in the hud cross is hard enough. I'm not a programmer so I don't know if this function can create?

Thank you very much for the active work OT you devote, hi. Pet

sthalik, 2017-08-14:
It doesn't work right. I tried it back in facetracknoir days. It keeps still, and then makes a sudden jump cause you made a larger movement. You can add `1.1` nonlinear sensitivity in the Accela filter with a similar result. Just don't go overboard, you'll have the same unwanted result if you drag the slider too far to the right.

petsild, 2017-08-14:
Such values I'm trying to keep.

AnY-cz, 2017-08-16:
this function could be dependent on the Accela filter. Change the value of the Z axis affected the percentage magnification smoothing or sensitivity.

closed 2017-12-02 by sthalik

Antonvrig123, 2020-03-17:
motion controls? how did you get it to work? I have tried to get it to work for a week with no success.