opentrack/opentrack #672: macOS use bundle instead of install command

MrCapone, 2017-08-07:
Current insatll script really outdated and even not works on my macOS Sierra. CMake can generate [MACOSX_BUNDLE](https://cmake.org/cmake/help/v3.0/prop_tgt/MACOSX_BUNDLE.html) including Info.plist, Icon, etc. I'm already working on creating .app bundle by current CMake tree. 
[image]
[image]
Do you think this is a good idea switch to MACOSX_BUNDLE and set old install script as derpecated?

sthalik, 2017-08-08:
Yes. Definitely! It's the Right Thing [tm] to do and much more forward-compatible. In fact there's no reason for the old script to remain after you're done.

We might wanna keep "install-fail-tool.sh" for editing dependency rpath though.

MrCapone, 2017-08-08:
@sthalik CMake can execute `install_name_tool` by itself without exernal scripts. But changing the current installation logic was too complicated. Ill try fix existing scripts for this now. And maybe later ill try make full CMake bundle logic.

sthalik, 2017-08-08:
@MrCapone empty `macosx/CMakeLists.txt` and write your logic in there. You can refer to `macosx/` as a full path using `${CMAKE_CURRENT_LIST_DIR}` and the install prefix as `${CMAKE_INSTALL_PREFIX}`.

Please don't change the minimal SDK version from `10.8`. I'm using Mavericks due to best Valgrind support.

Also, relative pathnames are in `cmake/opentrack-hier.cmake`. It's enough to change it for both the installation and the software at runtime to recognize the prefix. You have three blocks for Windows, Apple, and Unix specifying all relative paths in there. You can change all three to the same directory.

MrCapone, 2017-08-08:
As i said it more complicated. MACOSX_BUNDLE sets for executable target. So i need modify `opentrack-user-interface` target. Then i need create POST_BUILD custom commands for copying all stuff and executing install_name_tool. For now i made small fix for `install-fail-tool` https://github.com/opentrack/opentrack/pull/673 and was able build valid macOS bundle by 
```
# inside making subdirectory
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=./install ..
make
make all install
```
Default `CMAKE_INSTALL_PREFIX` sets as /usr/local but on my mac this directory not empty.

sthalik, 2018-01-04:
Feel free to return to the subject if you have the time.

closed 2018-01-04 by sthalik

MrCapone, 2018-01-25:
I kinda busy right now. But i remember about this feature. Also my track device broken and i need repair it for testing.

sthalik, 2018-01-25:
That's fine. We'll deal with it when you can.