opentrack/opentrack #674: Kinect tracker API

guilhermemtr, 2017-08-11:
Hi,

I've developed a [https://github.com/guilhermemtr/kinect-track-ir](head tracker for Kinect), and that I was hoping I could add it to open-track.
The main advantages of my tracker are:
  1. It uses Microsoft's Kinect API (v2) to track the face/head of the user, and after testing it for a while it seems much more accurate than, for example, facetracknoir.
  2. I've implemented everything with a publish subscribe mechanism, allowing to create pipelines to enhance the head tracker, making it more smooth, allowing it to dynamically adapt to the player's style of game during the game (without requiring any *a priori* setup). The publish subscriber design makes it extremely modular, and to make it work with a game it suffices to make a subscriber that interacts with the game (everything else can be left as is, unless you want to make changes to the data pipeline).

I've developed this tracker to play DCS, because I didn't want to spend 200£ on track-ir, but until now I haven't been able to make it work on DCS.
I've been testing the tracker by command line and it seems very accurate.

I'm sorry if this is not the right place to place this, but I didn't know where to put it.

sthalik, 2017-08-11:
It's right place.

Check out `tracker-test/`. You can copy and extend this stub to add another tracker.

The software builds with mingw-w64 and Visual Studio 2015 Update 3 and newer. You shouldn't have problems adding a tracker. If something happens, just ask.

guilhermemtr, 2017-08-11:
@sthalik thank you!
The thing is that it is not just a tracker.
It already has a pipeline of filters to make the tracking smoother.

The pipeline is implemented on top of a publish/subscribe system, where the tracker is the first publisher and each filter (except for the last one, which connects to some external listener) is both a publisher and a subscriber.

sthalik, 2017-08-12:
@guilhermemtr you can use the Qt user interface to allow users for setting up the filter chain. Otherwise you can return raw data to opentrack since it already has filtering and mapping functionality.

Maetharin, 2017-08-23:
Really interested in this in regards to using it either alone or in tandem with an EDtracker with the fusion tracker. Any ETA on when it will be added to Opentrack? No hurry, just curiosity.

sthalik, 2017-08-24:
There are no ETAs. You can ask for one if you contribute code and wait for review.

Maetharin, 2017-08-24:
I'm terribly sorry then if I caused any offense, just an avid fan here, though may I ask if this includes the Xbox 360 Kinect or just the Xbox One

guilhermemtr, 2017-08-24:
Hi, I just saw that they are not compatible (I don't own a 360, but I just saw that the Kinect API v2 is not compatible with the 360).

Well, it will take some time, and I'll only have time for this in January (at least) because I have two (VERY) important deadlines until December.
Anyway, if you really want it I'll have no problems in helping to incorporate it in opentrack.
I didn't do it because I've never developed for windows (usually I just program in Linux environments), and I didn't know how to test my API (other than reading what it outputs on the command line), that's why I came here, to try to plug it into opentrack (and that is pretty much what is missing, because other than that are only small tweaks, but those can only be made after it works).

PS: If there is a day in any weekend I don't work, I may look at it again, but before making it work on opentrack I still have to be able to compile and run the opentrack project, which I haven't done yet.

guilhermemtr, 2017-08-24:
(I'm sorry I closed this accidentally, not used to use issues, clicked the wrong button).

sthalik, 2017-08-25:
@guilhermemtr you already have your GNU environment on Windows. Use MSYS2 for mingw-w64 binaries for the whole toolchain. With CMake it'll generate build rules with no worries. Use the `Ninja` generator since the mingw-w64 port of `make(1)` has broken parallel compilation and possibly more. It's like `make(1)` except slightly faster.

I build with GNU and LLVM in addition to Microsoft's compiler, since the Unix toolchains are more strict about the language spec.

I can sympathise since I used to not know any win32-specific API at all. It all came around, and these days you can do all you want with portable code. The OS-specific layers are already there for you.

You can ask via my email or here. Don't feel bad about asking since people have been though the same procedures and know the answers already.

closed 2018-05-14 by sthalik

Slion, 2019-01-29:
Was such a Kinect v2 tracker ever completed?

Slion, 2019-05-18:
Open Track now supports Kinect:
https://github.com/opentrack/opentrack/wiki/Using-Microsoft-Kinect-V2

sthalik, 2019-05-20:
Please add yourself to README.md, AUTHORS.md and OPENTRACK-LICENSING.txt. Thanks!

Slion, 2019-05-24:
> Please add yourself to README.md, AUTHORS.md and OPENTRACK-LICENSING.txt. Thanks!
See #954