opentrack/opentrack #675: [Idea] Limit for relative translation

Maetharin, 2017-08-23:
Hello there!
First of all, thx for all the time you put into this, I highly appreciate it!

Now to my request:
I´ve recently discovered relative translation, making the fusion tracker experience of EDtracker and Aruco even more immersive. But whenever I try to just look around rotationally, the Aruco tracker picks up my rotation as poition change also. The result is whole lot of wobbling around.

Would it be possible to give the user the option to limit relative translation to a certain spin velocity? As in that it only works during slow rotation or after rotation has taken place.

sthalik, 2017-08-24:
Hey,

While this is a valid question, this isn't what you want for this problem.

- Temporarily select the Aruco tracker instead of the Fusion tracker.
- Click the 'calibrate' button in tracker settings while Aruco is tracking.
- Make lots of different poses: pitch AND yaw low, down, left, right, etc.
- Click the same button that was "calibrate". You should now get a custom Z position.

Your position data should be very close to zero. You can repeat the calibration procedure. It should show the dot as the center of your head. Remember that you must pitch and yaw at the same time for different poses to get registered in the statistical calibrator. Otherwise it doesn't have enough data to work with.

Maetharin, 2017-08-29:
Thank you for your help, it definitly made the whole affair smoother.
Though there still are two problem with relative translation.
First, the one relative to the original post, that my virtual head changes position during rotation.

Another issue is that for some reason whenever I try to move my real head along the x-axis while my virtual head is at an angle of ±90°, relative translation doesn´t reliably work. It does let me move my head around on the z-axis sporadically, causing wobble along the virtual x-axis as soon as I move my head out of that perfect sweet spot.

When I have my head at max angle ±30° or min angle ±120° it does work perfectly though.

Would you know how to fix that problem?

sthalik, 2017-08-30:
> First, the one relative to the original post, that my virtual head changes position during rotation.

This is what relative translation does. Nothing more. You can disable its effect on particular position axis or effect of rotation axis on all position axis.

> Another issue is that for some reason whenever I try to move my real head along the x-axis while my virtual head is at an angle of ±90°, relative translation doesn´t reliably work.

See if you can reproduce this in another sim.

Maetharin, 2017-09-02:
Concerning the first issue (or non-issue actually) would it entail a lot of work to introduce a certain limit as I asked for in the OP?
> Would it be possible to give the user the option to limit relative translation to a certain spin velocity? As in that it only works during slow rotation or after rotation has taken place.

It really is an issue that  f.e. when I look to the back over my right shoulder, I will be violantly moved to the left of the of the cockpit.

sthalik, 2017-09-07:
> It really is an issue that f.e. when I look to the back over my right shoulder, I will be violantly moved to the left of the of the cockpit when looking forward again.

It's the entire purpose of "relative translation" for it to behave like that. You can disable individual axis of rotation or position change.

sthalik, 2017-09-07:
Users want to look back, then move toward the side the rotated pilot's is naturally in. That's what this use-case is for. Relative translation is when you rotate first, reposition second.

You may want to disable it, and enable the "dynamic neck" feature. You can select its sentivity with the exact amount of neck displacement.

Maetharin, 2017-09-07:
Ah ok, never understood what that feature was about, thank you

closed 2017-09-07 by Maetharin