opentrack/opentrack #676: Linux. Opentrack (for linux) with IL-2 Forgotten Battles (wine).

ljgdasfhk, 2017-08-24:
I'm playing IL-2 Sturmovik Forgotten battles on linux (wine).
Whether it is possible to use the opentrack. How to setup?

sthalik, 2017-08-24:
Use the Wine protocol.

ljgdasfhk, 2017-08-24:
Tell me which items to choose in the CMAKE for use WineProtocol ?

ljgdasfhk, 2017-08-24:
In the output menu I see only the virtual joystick (EVDEV) and UDP

sthalik, 2017-08-24:
You need the `SDK_WINE_PREFIX` CMake cache variable set to your Wine installation prefix. For instance `/usr` or `/opt/wine`, depending on where you have it installed. You'll need the 32-bit multilib compiler. However, it works with both 64-bit and 32-bit Wine runtime directories. In case your Wine directory is different than `~/.wine`, you need to pass the `WINEPREFIX` environment variable also to opentrack.

ljgdasfhk, 2017-08-24:
[ 75%] Generating opentrack-wrapper-wine.exe.so
g++: error: unrecognized command line option ‘-std=c++14’
winegcc: g++ failed

ljgdasfhk, 2017-08-24:
[ 75%] Generating opentrack-wrapper-wine.exe.so
In file included from /home/someone/opentrack/proto-wine/opentrack-wrapper-wine-main.cxx:1:0:
/usr/include/c++/7/cerrno:41:10: fatal error: bits/c++config.h: Нет такого файла или каталога
 #include <bits/c++config.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
winegcc: g++ failed

gcc 7.1
wine 2.15

ljgdasfhk, 2017-08-24:
Can you write a list of dependencies for opentrack with WineProtocol ?

ljgdasfhk, 2017-08-24:
What is wine-protocol ?
How does he work?
Should I run "wine /programms/opentrack/libexec/opentrack/TrackIR.exe"   ?

sthalik, 2017-08-25:
No, don't run it manually. And it doesn't implement the protocol.

The Wine binary about to be compiled with `winegcc`, passes data from the Unix shared memory mapping to Wine's memory mapping. It's started automatically for the Wine protocol's runtime duration. That way Windows games can receive pose data to begin with.

You need a full multilib environment and `winegcc`. I recommend building Wine on your own just in case the binary package may be missing Wine headers etc.

```
g++: error: unrecognized command line option ‘-std=c++14’
winegcc: g++ failed
```

`winegcc` is a wrapper around `gcc`. It appears that your system compiler is pretty old. What's the gcc version?

```
#include <bits/c++config.h>
^~~~~~~~~~~~~~~~~~
compilation terminated.
```

Your Debian-based system splits files for multilib into packages that aren't installed by default. Use `apt-file` to find these packages and install them.

 > Can you write a list of dependencies for opentrack with WineProtocol ?

It's multilib gcc and winegcc.

sthalik, 2017-08-25:
What's your GNU glibc version? I may be able to publish a self-contained tarball. Just don't count on a `.deb` please. It's too much unnecessary work to make a single `.deb`.

ljgdasfhk, 2017-08-25:
I'm working on arch-Linux.
Now compilation works without problems.
What is the wine-protocol ?
Is it the protocol of the ®TrackIR ? What is the TrackIR.exe file for ?
What data is output by the opentrack with wine-protocol ?
Can I use them, for example, to control the mouse ?
How can I make sure that the opentrack outputs the data correctly, but the game does not accept them?

sthalik, 2017-08-25:
`trackir.exe` exists because some simulators fail to work with headtracking unless a process named like this exists. It doesn't do anything, just sits there.

The Wine protocol works with `NPClient.dll` and `NPClient64.dll`. For IL2-BoS you need `old_trackir = 0`. Some people had it set to 1 and had to change it back. It should work fine on a default installation.

You should be launching your Wine programs in a single prefix by one user only, in all (?) cases. Launching Wine by root is a sign you're doing something wrong.

Are you using one `WINEPREFIX` or multiple?

MrCapone, 2017-08-25:
I'm trying build Linux version with wine support too. And i got linker error 
```
ld: Relocatable linking with relocations from format elf64-x86-64 (/opt/wine-stable/lib64/wine/libwinecrt0.a(exe_entry.o)) to format elf32-i386 (opentrack-wrapper-wine.Kckdvs.o) is not supported
```
I was able build after removing `-m32` option at  [this line](https://github.com/opentrack/opentrack/blob/unstable/proto-wine/CMakeLists.txt#L17). Is this correct?

ljgdasfhk, 2017-08-25:
Now it became clearer.
I did not know about the need for NPClient.dll.
Probably it needs to be placed in the folder "/wine_programs/ IL-2-Forgotten-battles" and registered in the system.
I'll check it in 8 hours.
I use one prefix.

sthalik, 2017-08-25:
`NPClient.dll` path will be put in the registry by virtue of starting the Wine protocol. It's the game that loads the dll from the registry path.

sthalik, 2017-08-25:
> I was able build after removing -m32 option at this line. Is this correct? Does opentrack-wrapper-wine.exe need to be i386?

The error appears as if you had no 32-bit Wine-specific `crt0.o` which is a critical startup file for all gcc-produced binaries. If your installation is 64-bit only then it's correct. Otherwise it should fall back to multilib `libwinecrt0.a` for the 32-bit installation.

See if your prefix has other files named `libwinecrt0.a` and send me their paths.

MrCapone, 2017-08-25:
> See if your prefix has other files named libwinecrt0.a and send me their paths.
I have one at `/opt/wine-stable/lib/wine`
and other at `/opt/wine-stable/lib64`

sthalik, 2017-08-25:
so when you run `strace(1)` on `winegcc -m32`, it only tries to open the 64-bit archive?

MrCapone, 2017-08-25:
Yes. But i have 64bit wine in my system so it is normal.

sthalik, 2017-08-25:
Is it a dual wine64 wine32 install? It shouldn't happen.

ljgdasfhk, 2017-08-25:
Process il2fb.exe is 32 bits.
Which of the libraries loads the opentrack: NPClient.dll or NPClient64.dll ?

sthalik, 2017-08-25:
32-bit processes load `NPClient.dll`. You don't need to touch it. The simulators specifically load from a directory determined from the registry and not from the simulator's directory.

MrCapone, 2017-08-25:
I have wine 64 binaries but 32&64 ldev libs.

ljgdasfhk, 2017-08-25:
The opentrack does not register the library.
Something I did not understand.
After starting the opentrack, the path to the library should appear in the wine registry. But this does not happen.

sthalik, 2017-09-07:
We got this sorted out in #685

closed 2017-09-07 by sthalik