opentrack/opentrack #679: Missing make uninstall

Talkless, 2017-08-26:
When building 2.3.9 on  Kubuntu 16.04, I can call `sudo make install` to install opentrack into `/usr/local`, but there is no `make uninstall`:
```
sudo make uninstall
make: *** No rule to make target 'uninstall'.  Stop.

```

sthalik, 2017-08-27:
See <<https://cmake.org/Wiki/CMake_FAQ#Can_I_do_.22make_uninstall.22_with_CMake.3F>>.

Note that `install_manifest.txt` in the build directory is a plain text file. Use it like this:

```
% while read f; do rm -f "$f"; done
```

Talkless, 2017-08-27:
Aruco, Flightgear, OpenCV, OpenSceneGraph and who knows how many other projects has `cmake_uninstall.cmake` for this purpose, maybe opentrack could have it also?

sthalik, 2017-12-21:
It's a self-defeating scenario to rely on `make uninstall` for package management. Install to a separate prefix so you can nuke the whole directory tree.

You can send some cmake boilerplate code enabling the `uninstall` target, I'll accept it.

closed 2017-12-21 by sthalik