opentrack/opentrack #680: Does not link with aruco: undefined symbol: MarkerDetector::setMinMaxSizeEff

Talkless, 2017-08-26:
I'm building opentrack 2.3.9 on Kubuntu 16.04, with aruco 2.0.19. When I launch opentrack, I get this:
```
library "aruco" failed: "Cannot load library /home/vincas/code/opensource/opentrack/prefix/bin/../libexec/opentrack//libopentrack-tracker-aruco.so: (/home/vincas/code/opensource/opentrack/prefix/bin/../libexec/opentrack//libopentrack-tracker-aruco.so: undefined symbol: _ZN5aruco14MarkerDetector13setMinMaxSizeEff)"
```
There is no such function `MarkerDetector::setMinMaxSizeEff` in aruco, nor in 2.0.19 neither in older 1.3.0.

Talkless, 2017-08-26:
What should "-DSDK_ARUCO_LIBPATH" point to? Directly to .so, .a?

Talkless, 2017-08-26:
I have tried with .a (had to add `-fPIC` for aruco to make `libopentrack-tracker-aruco.so` link), and with .so, same issue.

Talkless, 2017-08-26:
It seems that `aruco-2.0.19/src/markerdetector.h` has this function defined inline, meanwhile there is a copy of headers in `tracker-aruco/include/markerdetector.h` and it has it declared out of line...

Why the headers are stored in `opentrack` repo anyway?

sthalik, 2017-08-27:
Clone this with `--recurse-submodules`:  <<https://github.com/opentrack/opentrack-depends>>.

Talkless, 2017-08-28:
Aruco from that repo worked, thanks.

It would be nice to have it documented more.. explicitly somehwere :) .

sthalik, 2017-08-30:
I agree it should. For now please ask :)

closed 2017-08-30 by sthalik