opentrack/opentrack #681: freetrack/opentrack greyed out calibration

Nickhurl7, 2017-08-27:
I have freetrack enhanced enabled. Using with TrackClip plus. Works great, except the calibration button is greyed out, so no calibration option, which I think may be producing some of the slightly odd issues I'm having, such as the camera picking up extraneous points of light behind my seating point.
Unless opentrack/freetrack enhanced doesn't need calibration when using TrackHat/Clip plus?

sthalik, 2017-08-28:
Start tracking, then calibration isn't greyed out anymore.

Nickhurl7, 2017-08-28:
Hi Stanislaw,

Thanks for the fast response. 
I don't get the calibration button 'active' even with tracking on. I promise I would not have asked the question if the solution was that simple. 
I tried reinstalling the software etc, but no go. The Track Clip Plus & Opentrack work perfectly other than this small issue with the calibration being greyed out. I also don't get the option to enter the Clips dimensions, but then again, the dimensions aren't provided on any of TrackHat/Clip Plus anyway, even on their website. 
I use the Clip Plus on the left side of my headphones. I also tried using the offsets & reverse & adjusting the various movement scales. No calibration still. 
I will send an enquiry to TrackHat as well, as this maybe a a feature of their latest hardware. Maybe it's self calibrating.

Thanks again

Nick

sthalik, 2017-08-30:
> I also don't get the option to enter the Clips dimensions

This is interesting. Show me a screenshot please.

Nothing is self-calibrating or self-setting-up.

closed 2017-12-02 by sthalik