opentrack/opentrack #682: headtracker.xml overrides Flightgear default offset

Talkless, 2017-08-28:
It seems that I have to set individual x/y/z offset in Opentrack UI for each aircraft I launch in Flightgear, to match default pilot's head center position...

It would be nice if opentrack would _add_ offset to the Flightgear offset (offset for the offset? :) ) that is currently applied, so if Opentrack sets Y to "0,10", this means that camera position would be 10 cm up from original pilots position defined by the aircraft authors.

Although I am afraid that this is feature request for the Flightgear itself, to have some soft of offset-2, for secold-layer offsets as mentioned, or maybe it is implemetable by NASAL in Opentrack protocol plugin..?

sthalik, 2017-08-30:
> Although I am afraid that this is feature request for the Flightgear itself, to have some soft of offset-2, for secold-layer offsets as mentioned, or maybe it is implemetable by NASAL in Opentrack protocol plugin..?

We're not using Nasal. We're binding the UDP data to fgfs's properties directly. I imagine Nasal would be more CPU-intensive here.

Also when I launched the default setup with and without connecting opentrack, the pilot's position was the same when centered.

Talkless, 2017-08-31:
Here are reports on Flightgear forum:
https://forum.flightgear.org/viewtopic.php?f=36&t=31117&p=316143&hilit=opentrack#p316143

> With the default cessna c172 I'm positioned right in the center of the plane (thus not on the pilot seat).

sthalik, 2017-09-01:
@Talkless if there's another property we can tie to, then sure. But saving the default property state and adding an offset is too brittle. It could be modified for an arbitrary reason. Also Nasal is overkill for this purpose.

Talkless, 2017-09-01:
> It could be modified for an arbitrary reason.

Yes, true. I have created topic for this issue:
https://forum.flightgear.org/viewtopic.php?f=6&t=32821
I haven't got any developer comment yet though...

Although, _if_ it would be possible somehow to implement it in NASAL _correctly_, please have in mind that there are probably hundred times more "stuff" being calculated by all these gauges in instrument panel, Canvas displays, etc., so I believe adding a few numbers wouldn't be a huge problem, compared to what's already done.

sthalik, 2017-09-01:
Please post here once there's a working solution.

Nasal can break if it's started after an offset's already established.

sthalik, 2017-09-01:
Also you can use custom center pose in opentrack as an ugly workaround for the duration.

Talkless, 2017-09-01:
Yes I am using Custom center pose. I put Opentrack window on top of FG and calibrate it accordingly for aircraft currently in use.

sthalik, 2017-09-01:
I think aircraft models should have the pilot position at offset zero. Not a fgfs content developer but still.

Talkless, 2017-09-02:
I have launched Cessna C172P, then open `View -> Adjust View Position`, and it shows -0.21m, 0.19m 0.36. 

A-10 has 0.00m, 1.50m and 3.20m.

This value is overridden by Opentrack when tracking is on.

sthalik, 2017-09-03:
Can opentrack know the default value?

Talkless, 2017-09-03:
I don't know that. That's more, pressing 'v' key changes view into "Model View", "Left Passenger View", etc, and these have their own different offsets...

sthalik, 2017-09-03:
There should be a separate offset for external headtracking control. I'll implement it once that exists.

sthalik, 2017-09-03:
Give me the name of stock aircraft with that issue please.

Talkless, 2017-09-03:
c172p for example.

Talkless, 2017-09-05:
I have started `flightgear-devel` thread, got one response, so I guess FG needs some changing:
https://sourceforge.net/p/flightgear/mailman/message/36024474/

sthalik, 2017-09-06:
I'll write some code for direct freetrack support once a solution's in place.

closed 2018-05-14 by sthalik