opentrack/opentrack #684: Freetrack Plugin issue with Elite Dangerous, but only in specific versions of Opentrack

c-f-warneke, 2017-08-29:
opentrack 2.3.1.4 and opentrack 2.3.9's Freetrack Plugin seems broken for Elite Dangerous while 2.3.7 works fine. For reference:  any of the mentioned versions works fine with Warthunder. I just wanted to give you guys a heads up on that , since the behavior seems specific to the opentrack version and may indicate a larger problem.

MrCapone, 2017-08-30:
Im using opentrack 2.3.9 (built by myself from unstable) with freetrack protocol on 2 games. I have Elite Dangerous ion my steam but it not installed right now. I'll test it later.

sthalik, 2017-08-30:
Did you install opentrack on the same drive as ED? Otherwise it won't work. This is an ED problem.

c-f-warneke, 2017-08-30:
well I now managed to brick support for elite completely, first I tried to copy my 2.3.7 install from d:\ to e:\ to test if it would stop working since elite is on d:\, but then after copying it back it still didn't work I then uninstalled and reinstalled via the installer and: nope totally bricked now. that's weird.

MaZyGer, 2017-08-30:
Ok I had all the time same problem. ED was not working but other games like Projects Cars was working.. My ED Is on F:  and I always install on F (default settings in my systems). So now I thought I try on C drive... yes there it works.. 
My path is now C:\opentrack.

c-f-warneke, 2017-08-30:
I must agree with mazyger, sthalik. I can acutally use opentrack with ed fine, but now only if its on c:\ . I deleted all registry keys relating to opentrack and npclient.dll. Now I can even use 2.3.9 but seemingly from c:\ only. Now I am completely confused.

sthalik, 2017-08-31:
@c-f-warneke 

> well I now managed to brick support for elite completely, first I tried to copy my 2.3.7 install from d:\ to e:\ to test if it would stop working since elite is on d:\ , but then after copying it back it still didn't wor

You must start tracking once before starting the game in order for the path to be updated. Otherwise the game sees the old nonexistent path.

@MaZyGer 
> When I tried even in the ED folder it didn't work

No.

My installation directory is `d:/Steam/blah/ED`. My opentrack installation directory is `d:/dev/opentrack/build/install`.

Install it on the same drive, start tracking once and it's done.

Also no need to use the installer but that's just me.

@c-f-warneke 
> I must agree with mazyger, sthalik. I can acutally use opentrack with ed even if it's not on the same drive as ed, but now only if its on c:\ . I deleted all registry keys relating to opentrack and npclient.dll. Now I can even use 2.3.9, but again from c:\ only. I am completely confused.

So am I. It started on some early ED patch. You may want to check if running both opentrack and ED as administrator will help. If so, disable UAC entirely via Winaero or Group Policy. Otherwise we're back to square one.

c-f-warneke, 2017-08-31:
Tried it with administrator account. Same behavior, only when its on c:\ elite will hook freetrack.

c-f-warneke, 2017-08-31:
Oddly enough I cant reproduce what I originally observed with freetrack working on d:\.

c-f-warneke, 2017-08-31:
So, I am back with results: I put opentrack 2.3.9 on D:\ but also put a copy of npclient64.dll in c:\opentrack (folder otherwise empty). I then started tracking and after that used the locate dll option to change the freetrack reg-keys to point to the one npclient64.dll-file I put  on c:\ and as you would expect, elite worked. So yeah seems to definitely be an issue with elite not swallowing reg keys that point to non-c:\ drives. 
So, what I now wonder, could it be that freetrack not always correctly updates the regkeys? So that in mine and maybe sthalik's cases it only worked on drives other than c:\ because actually there are copies of npclient on c:\ (in my case that would have been true) and the reg keys just still pointed at the files in old installations on c:\?

c-f-warneke, 2017-08-31:
Addendum to my former post:  Just to be sure I deleted npclient.dll and npclient64.dll from the opentrack directory on my d:\ drive while testing.

sthalik, 2017-09-01:
It updates the registry keys unconditionally.

Please use `sysinternals procmon` to find the filename ED tries to open, if at all. I've had reports that it doesn't try to open pathnames at all, if it doesn't like them.

closed 2017-12-02 by sthalik