opentrack/opentrack #685: About Wine protocol

ljgdasfhk, 2017-08-29:
> NPClient.dll path will be put in the registry by virtue of starting the Wine protocol

I run the program with a command
`lxqt-sudo env WINEPREFIX=/home/user/.wine /programms/opentrack/bin/opentrack`
Then I search the registry for a string "npclient". She is not there.

What is the TrackIR.exe file for ?
> trackir.exe exists because some simulators fail to work with headtracking unless a process named like this exists
Should I run "wine /programms/opentrack/libexec/opentrack/TrackIR.exe" ?

sthalik, 2017-08-30:
> Then I search the registry for a string "npclient". She is not there.

See <<https://github.com/opentrack/opentrack/blob/unstable/proto-wine/opentrack-wrapper-wine-windows.cxx#L45>>.

It only happens once you start with the Wine protocol selected. Also you should have Wine running with opentrack's wrapper once it's started. The process name without extension is opentrack-wrapper-wine <<https://github.com/opentrack/opentrack/blob/unstable/proto-wine/CMakeLists.txt#L17>>.

And don't run opentrack as root. Otherwise I'm assuming running as root is the reason it doesn't work.

> Should I run "wine /programms/opentrack/libexec/opentrack/TrackIR.exe" ?

For starters, run client.exe from here: <<https://github.com/opentrack/opentrack/tree/unstable/contrib/trackir-client>>.

It's a Windows console program so run it under the terminal. See if data pops out.

ljgdasfhk, 2017-08-30:
> Also you should have Wine running with opentrack's wrapper once it's started

What exactly should I run? `wine ./opentrack-wrapper-wine.exe.so`   ?

If I run the program on behalf of a regular user, I get a message :  ≈ one of the libraries did not load

sthalik, 2017-08-31:
@ljgdasfhk 

> What exactly should I run? wine ./opentrack-wrapper-wine.exe.so ?

It's gonna run on its own when you start tracking.

> protocol load failure

It's because you created the shared memory mapping as root. Delete it manually (using Perl for instance) or just reboot. The only check is that the shared memory mapping can be created and accessed.

ljgdasfhk, 2017-08-31:
Suppose I rebooted and ran the program `$ ./opentrack`
 Will the shared memory mapping work well?

sthalik, 2017-08-31:
Yes.

ljgdasfhk, 2017-08-31:
The wineserver starts when the operating system starts by root. Could this be the problem?

sthalik, 2017-08-31:
Don't use your own WINEPREFIX.

ljgdasfhk, 2017-08-31:
And whose prefix to use ? Should I Create a new one? What's the difference?

ljgdasfhk, 2017-08-31:
Create new WINEPREFIX:
`WINEPREFIX="/home/user/.wine_32" WINEARCH=win32 wine wineboot`
Start opentrack:
`env WINEPREFIX="/home/user/.wine_32/" /programms/opentrack/bin/opentrack`
Start with wine-protocol
Everything works great !
THX !

sthalik, 2017-09-01:
@ljgdasfhk your very last post has it all correct.

sthalik, 2017-09-01:
Though @ljgdasfhk running Wine as root is ALWAYS incorrect.

closed 2017-09-01 by sthalik