opentrack/opentrack #688: 2.3.10 development process

sthalik, 2017-09-23:
Hey,

I since started work on 2.3.10. Let's outline some of my features and ideas:

- [x] PT tracker: different modes for floppy filters, red-only LEDs<sup>(#note1)</sup>.
- [x] Octopus: no longer has artifacts at the bottom or elsewhere
- [x] PT tracker: start amcap for exposure control for the PS3 Eye.
- [ ] PT tracker: we could make amcap automatically open the filter settings window for given display mode (frame size, fps) and close once it's done.
- [ ] Support for the Tobii EyeX devices (not started, planned out, time-intensive).
- [ ] Support for joystick input on Linux and OSX via QtGamepad (not started, very easy, possible third-party contribution)
- [ ] Notice of obsolescence for Windows XP support in the near future.
- [ ] Notice of obsolence for CPUs without SSE3 support in the near future. All Intel and AMD chips since 2004 support SSE3.

- - -
<a name="note1">(1)</a> Floppy filter mode ignores sepia-spectrum colors as from sunlight. I'll post a teaser but today it's been cloudy all day and artificial light can't show the effect. The mode uses 80% of the blue value and 20% of the green value to establish the grayscale color. Normally [Rec. 709](https://en.wikipedia.org/wiki/Rec._709) is used for grayscale conversion which was already complained about.

Nothing prevents us from adding other extraction modes for e.g. green or blue-colored LEDs. The "80%" formula will probably change.

Red only:
[image]

Blue/green:
[image]
- - -

albertolg, 2017-09-24:
is too hard to implement a 3d head like was in freetrack instead of the octopus and disable it when freetrack is minimized to save the cpu ?

sthalik, 2017-09-24:
It's time-consuming. I personally have no interest in this. It's an improvement however, and if someone sends the code, I will accept it.

Len62, 2017-09-27:
So with amcap implemented we would no longer need the cleye driver, as far as OT is concerned ?

sthalik, 2017-09-27:
It's not a driver. It's an app that shows camera options successfully.

Len62, 2017-09-27:
Got it, thanks!

sthalik, 2017-09-28:
Replacing the CL-Eye driver is by all means possible -- Linux has the right byte sequences to emit in its own driver. Someone could reuse those in a WinUSB driver. But it's a lot more work than finally-working filter options.

Len62, 2017-09-28:
Having working filter options will be great. I generally do not have to change mine but for others it will be nice to be able to adjust options in OT.

sthalik, 2017-09-28:
I found something strange. See <<http://www.driverscape.com/download/usb-camera-b4.09.24.1>>. The **BOTTOM** link. Top is some adware. Is that a free ps3 eye driver?

Len62, 2017-09-28:
The cleye driver is version 5.3.0.341, device ID# reported for windows is usb-camera-b4.09.24.1. Looks like an earlier driver version to me.

sthalik, 2017-09-28:
Yeah, but free and aparently legal.

Len62, 2017-09-28:
Free is always good! I may give it a try just to see if there are any big differences.

albertolg, 2017-09-28:
@sthalik I found this https://github.com/HipsterSloth/psmove-unity5/wiki/PSEye-Software-Setup-%28Windows%29

the camera is working, but opentrack don't see it

guillerodriguezv, 2017-09-29:
Hi, excellent work,  may I ask for Pimax support in your next release,  please. 

Thanks

albertolg, 2017-10-02:
@sthalik it's possible to use the libusb drivers with the ps3eye so we can get rid of the old codelabs drivers?

sthalik, 2017-10-02:
@albertolg yes, and I'm looking at libusbK specifically. The API isn't braindead unlike Microsoft's. It can be used inside a custom DirectShow filter for the webcam.

albertolg, 2017-10-02:
great news @sthalik thanks

sthalik, 2017-10-06:
I'm gonna add it and disable by default.

Also, I added "portable installation" mode. Creating "portable.txt" in the software's directory turns it on. Configs are gonna get created in the program's directory.

Len62, 2017-10-06:
Sounds good. Auto update to only release versions, correct?

sthalik, 2017-10-09:
Yes of course.

sthalik, 2017-10-13:
You should like this:

[image]
[image]

petsild, 2017-10-13:
Yeah awesome

Len62, 2017-10-13:
Very nice!

503Kento, 2017-10-13:
Agreed, very nice, should help with my uneven LED brightness that I'm too lazy to fix.

sthalik, 2017-10-13:
Grab it as usual from <<http://opentrack.misaki.pl>>.

Len62, 2017-10-14:
Test build performs really great. It was nice to be able to use amcap to get to the ps3 eye camera settings, and the color channel options are a great addition. The changes to the point extraction settings will make it easier for users to find that "sweet-spot".

sthalik, 2017-10-18:
@Len62 I intend to make the `amcap` dialog automatically use the right camera and its settings, pop up the filter options and exit when they're closed. It's just an old app and very annoying to deal with. Simply disabling options like MPEG2 capture, sound, saved settings, etc. took way more than it should.

Also we made a Discord channel with @mrsanchos. You can drop by to discuss development if you want. The link -- <<https://discord.gg/dNS2hhB>>.

MathijsG, 2017-10-25:
Somehow amcap isn't opening here with the test build. Can I debug something for you?

Len62, 2017-10-26:
Doesn't open at all, or screen is blank? If OT is running then the amcap window opens w/blank screen. If you stop OT then amcap opens correctly, at least for me, on the newest test build.

MathijsG, 2017-10-26:
Nothing opens at all. That's the weird thing. While the separated executable in the same test folder of latest Opentrack test, does open.

Len62, 2017-10-26:
Weird for sure. @sthalik will probably have an idea.

sthalik, 2017-10-29:
@MathijsG,

These aren't solutions but possible means of identifying the problem

- did antivirus pop up? Avast is known to whine about webcam usage
- check Event Viewer

sthalik, 2017-10-29:
Can you check if my blog post is understandable <<https://sthalik.blogspot.com/2017/10/windows-timer-resolution-nuclear-option.html>> for non-programmers? Is it too verbose and without context? Thanks!

sthalik, 2017-10-29:
I posted `opentrack-test-20171029_01-BACKUP_YOUR_CONFIGS.7z` to <<http://opentrack.misaki.pl>>.

**BACK UP YOUR CONFIGS**
**BACK UP YOUR CONFIGS**
**BACK UP YOUR CONFIGS**

albertolg, 2017-10-30:
@sthalik any news about using libusbK instead of cl-eye-drivers? thanks

sthalik, 2017-10-30:
@albertolg it's very hard compared to regular opentrack stuff. Don't wait on it.

Len62, 2017-10-30:
Tried out the latest test build, backed-up my configs, and everything ran smoothly. The octopus seems a little more fluid, did not run into any problems with the config files. Even running the previous test version w/a config from the newest worked good. Will build a new config from scratch later on today for the new test version. Read your blog and I don't think non-programmers would find it too difficult to understand. I would probably get rid of the "F" word, makes it seem somewhat juvenile, I'm not offended by it, I spent 22 years in the military. Got a real kick out of seeing the assembler code, I had learned a little myself a long time ago, but then life got in the way and I gave it up. I wished I hadn't.

sthalik, 2017-10-30:
The threshold slider settings aren't passed onto the old version. It only migrates old -> new. Just so you know.

Do you have any problems with shortcuts, either joystick or keyboard? Now it's ok to press a shortcut very quickly, it will register. But code changes break stuff as you've seen through the project's history.

> I would probably get rid of the "F" word, makes it seem somewhat juvenile

Thanks, you have a point. In Polish one can compose a sentence with 80% profanity and still transmit plenty of information. In English swearing looks lame, kinda.

MathijsG, 2017-10-30:
@sthalik I understand your blog post, but I do see myself as having some knowledge of programming. However I miss the underlying problem. First you introduce the lack of resolution as a problem and then you jump to issues with Spotify. I'm assuming Spotify is using this 'fuck system performance' workaround you are mentioning? And you trying to undo that with your own DLL? Or am I totally wrong :-P.

Len62, 2017-10-30:
I use two shortcuts - toggle = ctrl + stick button, center = shift + stick button, and have not noticed any issues. Sometimes w/OT running, and tracker off, if I start a game and then toggle the tracker on the game will lock-up. I figured the threshold settings were not passed back, but it did not have any adverse effect on the previous test version.

MathijsG, 2017-10-30:
By the way is it possible to upload regular zip instead of 7-zip's own fileformat? Wanting to try this version on other laptop, but I end up having annoyances with Windows 10 apps for opening a .7zip file without having to bloat my laptop :-P.

sthalik, 2017-10-30:
@Len62,

> Sometimes w/OT running, and tracker off, if I start a game and then toggle the tracker on the game will lock-up.

Looks like a logic error in the game's head tracking support. opentrack's code inside the game process is so simple it doesn't even treat having head tracking on or off separately. There's the matter of particular .csv game protocol version, but it's handled exactly once and in the same way, inside the game's process' opentrack library.

@MathijsG,

> By the way is it possible to upload regular zip instead of 7-zip's own fileformat?

Damn, 7-zip is the first thing I install, even before notepad++ :) You won't get anywhere without an archiver.

Len62, 2017-10-30:
I did not think the hot-key issue was related to OT, more to the game. I just start the tracker before starting the game and then there are no issues w/the hot-keys. I thought about looking at Python just for the heck of it, don't know that I have the patience for programming nowadays.

MathijsG, 2017-10-30:
Interesting: When I drop the test folder directly in C:\ it works perfectly, but when I drop it on my Desktop, which makes it E:\Users\Mathijs Groothuis\Desktop\ it doesn't open.

Maybe it's because of the spaces? Or maybe something that assumes everyone is working on C:\?

Len62, 2017-10-31:
Just noticed that in the newest test version that the modifier keys are not being seen when setting hot-keys.

sthalik, 2017-10-31:
@Len62 great find! I updated <<http://opentrack.misaki.pl>> and removed the build from two days ago. There are no other changes so it's the old build + the modifier fix.

Don't play Telltale's Game of Thrones! It's basically a humiliation simulator going on for 5 DLCs. No matter what you do in any story arcs, you lose. It should be a 5-minute game with the player being presented a game over screen right after the game starts. Especially don't pay for that shit.

Len62, 2017-10-31:
Thanks! D/L'd the new test version but will have to wait to the morning to try it out. Have grandkids to take trick or treating tonight.  I only play two simulations, IL2-BOS and DCS World (mainly for the helos).

sthalik, 2017-12-26:
Please check the `20171226` build on http://opentrack.misaki.pl.

#### Arma anticheat notes

For Arma anticheat use the .dll's in `arma-anticheat/` directory. Copy to `modules/` and overwrite. Otherwise PLEASE test the new dlls. They should free up mouse view when tracking is stopped. That's the purpose of the new .dll's.

Only after they're fully tested I'm sending for whitelisting.

I think we should go for `2.4.0` as the version number since so much has changed already.

### Important:

The freetrack npclient `.dll`'s need your heavy testing with each flight game you have. please test each game. the new feature is freeing up mouse control when opentrack is stopped/closed.

Please check for bugs in old functionality since internal changes could've broken things that worked.

I'm really tired and weak. I'm looking for a co-maintainer. I'd also not like the project to die because of my possible permanent inactivity. I'm looking for someone with a few years of hobby/work experience and some C++11 and up knowledge, or old-style C++ knowledge and willing to learn the new-style C++. We're using a subset of C++17 features.

In the past, people with decent experience like @DaMichel found their way around the codebase easily. You can too if you want.

### Changes:

- dialog with the reason shown in case a protocol/tracker/filter can't initialize. Not used for vjoystick or realsense cause they have better dialogs than a stock error window.
- heavy refactorings to make former possible
- other heavy refactorings to make codebase look better for programmers
- reorg settings dialog. comment if it got worse.
- speedup tracker/pt color to grayscale conversion
- default to natural grayscaling in tracker/pt since it's way fastest
- allow freetrack mouse control when disabled, see beginning of this message
- make spline snap value lower for translation mappings

sthalik, 2017-12-26:
For testing freetrack with existing games, please:

- check with opentrack not started first
- stop opentrack, see if it resumes tracking after it's started again
- see if mouse control returns when starting again

Please list games you checked.

MathijsG, 2017-12-26:
Will test tomorrow with the games I own. Hope you will get better soon mate. Sounds serious.  Unfortunately I'm not really a programmer, more of a UX and bug squatter.

sthalik, 2017-12-26:
By the way, @MathijsG thank you for dozens of contributions you've made over the past years.

Len62, 2017-12-26:
Will give this a test later today. Sorry to hear about your health issues, hoping you get better.

MathijsG, 2017-12-27:
Euro Truck Simulator 2 not working. Will investigate further.

Len62, 2017-12-27:
I only play IL2-BOS and DCS World, no issues with either one.

MathijsG, 2017-12-27:
Somehow it didn't work with any program (FaceTrackNoIR, Trackhat, etc) so I restarted the game, then it worked perfectly. No issues with ETS2. Camera settings problem still exists.

sthalik, 2017-12-28:
What camera settings issue?

MathijsG, 2017-12-28:
See https://github.com/opentrack/opentrack/issues/688#issuecomment-339672961

The Camera Settings app doesn't open Amcap for my PS3 Eye camera. It just doesn't open anything.

When I put it in "C:\" everything does work. Also the amcap.exe executable does work seperately. Looks like something with spaces in the path maybe?

sthalik, 2018-01-04:
@MathijsG I can verify it's a bug. There are missing quotation marks around the executable path.

But amcap needs a bit of work to select the right camera, resolution, fps, open the filter dialog and close once it's closed. I'm not done with this fully yet.

@MathijsG @Len62 have you checked whether your games track after you stop and restart tracking? Also it should return mouse control while it's disabled.

sthalik, 2018-01-04:
I FIXED THE PS3 EYE CAMERA SETTINGS. IT WORKS FULLY WITHOUT AMCAP.

Len62, 2018-01-04:
I did not have any issues stopping and restarting. I'm going to try the latest test build in a little while, and see how things go.

MathijsG, 2018-01-04:
No issues yet with ETS2, will try more games when I am avoiding writing my thesis :-P. However the only 'issue' I encounter with ETS2 for example (and I remember more games in the past had this behavior) is that when you start Opentrack (I mean open the application) AFTER the game has been launched already you sometimes can't get the tracking to work in-game (the octopus all works and stuff) but this is also the case with the current version of Opentrack, Trackhat and FaceTrackNoIR, so maybe some games only load certain modules when they detect something TrackIR-like present on the system?

[57 more replies]

closed 2018-08-04 by sthalik