opentrack/opentrack #689: Some weird ATS opentrack/track IR conflict

SkeleCrafter, 2017-09-27:
It seems this issue has happened many times to other Euro Truck and American Truck Sim users but mouse-look is not working because opentrack is installed. I have no clue why. Editing the config.cfg folder for American truck simulator disabling g_trackir is not working because when you restart the game it just flicks it back on. When I check out the game log it shows this:

> 00:00:00.482s : uset g_trackir "1" 
00:00:00.719s` : [TrackIR] Device initialized. TrackIR software version 5.00.

Is this a ATS bug or a opentrack issue? Many other forums state uninstalling opentrack will remove this issue but opentrack isn't even formally installed on my system. It's just sitting in my downloads folder. So maybe you or anyone else has an idea of this issue?

SkeleCrafter, 2017-09-27:
Found the issue. I had two simultaneous copies of ATS. Removed now. Technically this issue is irrelevant to opentrack after all so... deleting this issue would be a good idea.

closed 2017-09-27 by SkeleCrafter

sthalik, 2017-09-27:
In the future, go to freetrack protocol options. Then select "disable on program exit".