opentrack/opentrack #691: Tracker + PS3 Eye track data all over the place.

Zubkover, 2017-10-04:
Hi,

sooo, I have a slight issue? Namely my darn face doesn't want to be tracked. Instead it tracks EVERYTHING beside it - even a darn yogurt cup. When on one-tone background, no tracker info is captured.
Aaaand it's on every filter. I did play with point extraction settings and filtering out the noise (since on stock settings there were freakin' thousand points jitterring all across the capture (especially in corners), but when I got rid of those then still - no info on my position.

Any ideas?

P.S. Rodaku, będę ja wdzięczny bardzo za pomoc! ^^

Zubkover, 2017-10-04:
Ow and just so we'd be clear - so far going thru the topics at this page I didn't notice any bold statement about the need of hacking the cam with the removal of IR filter, right?

Also - from test image thru drivers the image is fine, just the way it should be, but in opentrack it becomes purple-black and rotated every way possible.

Also If I were to switch to IR emitters stuck to my headphones... is there any specific req. or the regular tv remote IR ones will do?

sthalik, 2017-10-04:
> Also - from test image thru drivers the image is fine, just the way it should be, but in opentrack it becomes purple-black and rotated every way possible.

Try capturing at the same framerate, i.e. your USB controlled can't handle the required bandwidth for the device.

> so far going thru the topics at this page I didn't notice any bold statement about the need of hacking the cam with the removal of IR filter, right?

You don't need to. The aruco module tracks printed paper marks.

Zubkover, 2017-10-04:
But... on the preview it works with 75fps mode? And doesn't with 30fps in opentrack?

Yea, but I find IR more... let's say more of a plan. I'm living near Wilson in Poz, so I have more than enoguh shops with electronic here to do these in one hour or so... AND I DON'T HAVE A FREAKIN PRINTER. XD
So that's a problem xD

It apparently tracked my headphones in the dark... BUT it was SO OFF that it wasn't even worth it. xD Mainly cause filter detects and immediately discards track points, so the whole program is acting off. ^^

sthalik, 2017-10-04:
You live near Park Wilsona in Poznań, did I get that right? I love this city, impossible to get lost.

So a couple of hints with your LED models:

- people making that particular clip shape and not some other, arbitrary shape is meaningful. Due to some math & geometry some models work and some don't.
- try making a clip model that doesn't let either the upper or lower LED cover the middle. You don't want them to get swapped vertically. Give it enough space for pitch up and down.
- stick some unexposed film or a floppy, cut out into a circle, in between the PS3 Eye's lens and its black plastic surrounding. It dims the visible light spectrum a lot.

I have some work in progress on blue+green-only filtering. Not enough for a new release. Sadly there are barely any sunny days so I can't check how well it filters out sunlight.

Zubkover, 2017-10-05:
I also heard something about the "removing IR filter" mod? Is that a thing in this use case?
Also - still for the IR thing - ANY IR emitter itself will do, even those salvaged from tv remotes and such?

And if I do go that route I still do the Point tracker + freetrack combo?

Len62, 2017-10-05:
If you plan on using IR leds, then yes, you should remove the IR filter from the camera, then add the piece of floppy/unexposed film. AFAIK the IR leds from remote devices should work.

Zubkover, 2017-10-05:
I actually tested these, cause I remembered about having A GOSH DARN LED IR REMOTE  (XD) at the side of my desk - it detects it even without the IR filter removal o.o
And it's definitely a zener diode, cuz I can see no visible light from it. ^^

SoulMaril, 2017-10-10:
Are you sure a zener diode emits light?
What camera ara you using?
What type of clip you have?

Make sure you set the "Point Extraction" 
-Automatic Threshold "Unchecked"
-Threshold is around %70
-Min size 1px
-Max size 50px

I am using a ps3 eye cam with 3 point hat clip with white LEDs with above settings even in day time wihtout any issues. I do not use IR leds, I did not modified my cam in order to remove IR filter. I do not use any kind of filters to filter ambient light. Just the above settings. 

Also make sure you haveadjusted the camera exposure properly (video capture filter). Do not use any auto settings on the camera driver.

sthalik, 2017-10-10:
@SoulMaril he's having USB port bandwidth issues. When there's dropped data, the image gets malformed.

sthalik, 2017-10-10:
@Zubkover reduce the framerate, even to 30 FPS, in order to see whether the USB controller is dropping frames. If the frames stop being deformed at lower FPS then it's the controller's fault.

Zubkover, 2017-10-10:
But the thing is that it isn't. It actually smoothes out when I get to higher set Hz. And I have native 75Hz panels so I do see a difference.

Ok. So. I have mounted the LEDs as of now BUT it's still shietty as heck. The filter often switches from one diode to the next as center point causing my fov to fly all over the place without any consent.
And below I'll show ya what I mean.

And @SoulMaril if you can't see something it doesn't mean that it doesn't exist. ^^
[image]

Zubkover, 2017-10-10:
[image]

So yeaa. https://youtu.be/SOTNofgV4Aw
It works like this even when I replaced the burned diode of the left. ^^'

sthalik, 2017-10-17:
Does this happen only in opentrack?

Zubkover, 2017-10-20:
As seen above. All the cam recording and capture programs like OBS, Skype, ManyCams and CLDriver itself is seeing everything in order. Only opentrack has these issues. :C

sthalik, 2017-10-20:
Now that's a key piece of information!

Did you install any codec packs? Try it on a different PC, one you didn't install the same codec pack on.

Len62, 2017-10-20:
What does the video look like in OT when you use the 3 pt clip?

Zubkover, 2017-10-20:
Beside K-Lite and proprietary one for decoding Sony Alpha footage not really...

In the OT on 3pt clip is just basically getting all the info about LEDs by tracker, but some of them randomly fading to <0,9px and thus tracker switching to other diode as center point.

Zubkover, 2017-10-20:
Vid of it: https://youtu.be/OrAYmdhHVVs
I suspect that the jitter is caused by that switching... I guess at the very least.Threshold of detection cannot be adjusted any higher, since A HELLA LOT of noise is being detected then, so any more than 3 is passe. Same with min pixel size. Below 0.9px it's too much.

Len62, 2017-10-20:
First thing I noticed is that you have not calibrated your set-up. Once calibrated you will have a yellow cross that indicates the approximate center of movement, it is separate from the LED spots. Also, holding the leds between your fingers probably doesn't help due to hand movement, plus the LEDs are really close together which can cause issues. Can you mount them to a clip, or hat ?  With a 3 point clip my view looks this way:
[image]

sthalik, 2017-10-21:
@Zubkover Is this a cap or a clip? Since you have the center indicator jumping between LEDs then the PT tracker isn't able to detect your model properly.

You need to wear the model upside down if your video feed is upside down. Once it's detected properly, the yellow cross will hold firmly on the middle LED.

Zubkover, 2017-10-21:
Len62 Yea, I noticed that in set-up vids too, but no matter if taped to my headset or not - it's the same.

Sthalik - It's no more than litterally some wires, 47Ohm resistor, 3 IR diodes and USB plug. ^^
And this makes perfectly more sense xD But I have to revert all the jaw pitch and etc. settings too, right?

sthalik, 2017-10-21:
Revert what and why?

Zubkover, 2017-10-21:
*tfu* Not revert. I ment to write "rotate" XD

Zubkover, 2017-10-21:
Well. Either way that helped litterally with nothing. xD
It detects one LED as multiple points, white the other one is just going in and out, despite being detected at 1.2px+ and holding somewhat stable... so...
The question is.. should I just go for some aftermarket option (pref. something from PL or aliexpress) or just fok it, since it clearly doesn't want to work in my setup? xD

Zubkover, 2017-10-21:
https://youtu.be/HQxWZRWaszg

sthalik, 2017-10-21:
1) increase exposure to max, gain to middle value and color correction all to the same value.
2) increase gain until the points start blooming out
3) if they won't, decrease fps a bit

I also have a very special build that blurs the points but I hope it won't be necessary for you.

sthalik, 2017-10-21:
None of gain, exposure, color correction options should be set to auto.

Zubkover, 2017-10-21:
@sthalik But where can I access these options though? I'd love an SS to see for myself. ;)

sthalik, 2017-10-21:
CL-Eye Test.

Zubkover, 2017-10-22:
Well... Not really much has changed. Still tracker either looses points or multiplies one diode as many, and cross is either still on middle one or jumping from one to another.
And image is still purple and inverted. :C

Zubkover, 2017-10-22:
@sthalik Are you still in Poznań by any chance?

sthalik, 2017-10-22:
@Zubkover I live here. Why do you ask?

Zubkover, 2017-10-22:
Maybe you could come by and troubleshoot? I could repay you with a typical 4-pack of your liking or whatever floats your boat...

I'm asking since to the point where I would replace diodes/PS3Eye I won't know what is actually wrong in here. And my future plans of using phone as VR HMD (+ VorpX) still has to utilize opentrack, since from what I read  It cannot get gyro data "just like that".

But don't feel pressured in any shape or form. If you'll say that you don't really care It'd be perfectly understandable. ;)

sthalik, 2017-10-23:
@Zubkover install Windows in a VM and forward the webcam there, with VirtualBox or VMware. A fresh install will have all codecs in a working shape. Unfortunately I don't know a better way to fix codecs than to reinstall. But at least that way you can check.

Fuck the LEDs, as long as you don't short the wires. Just isolate them, and use grounding for the PC since it's the PC that's responsible for USB grounding.

Thanks, but no need to buy me any beers. Sadly I don't know any better solution but to reinstall Windows. You can confirm it with the aforementioned VM.

Zubkover, 2017-10-23:
> Fuck the LEDs
XD
I like where this is going. xD

I have one more way of checking it... but it'll have to wait a good amount of time. D:

I'll come back when I do that.

Zubkover, 2017-11-14:
So. On another system with another board the camera is fine, but still no matter the gain/exposure settings it like "flashes" which doesn't seem healthy. xD

sthalik, 2017-11-15:
I can't help you with that. Out of 3 PS3 Eye webcams I had, none had these symptoms.

Zubkover, 2017-11-16:
I mean like... it still is natural with EyeTest, OBS capture, Skype, but opentrack is the one that's not really coping with it particularly well. D:

Zubkover, 2017-11-24:
@sthalik Quick "modified" question - If I were to use phone + cardboard which tracking setting should I use? Is there anything that enables gyroscope adaptation?

sthalik, 2018-01-04:
@Zubkover "gyroscope adaptation"? Use "FreePIE IMU" for Android.

closed 2020-01-27 by sthalik