opentrack/opentrack #692: Opentrack Preview FPS

rinsatomi, 2017-10-16:
Hello, recently I installed the latest opentrack 2.3.9 with the PSEye camera and latest CL-Eye drivers. The problem is that even though I set it to 60fps the preview window is locked at ~30fps.

[image]

Is this reason for concern since at the Camera Info screen it displays the proper amount of frames per second? Should I rely on the FPS info? Thanks in advance.

[image]

sthalik, 2017-10-16:
You've got a good eye, it's *exactly* 15 fps.

It's limited on purpose. The purpose of the video feed is checking whether the capture works. It doesn't look too awfully choppy but displaying at full framerate would take unnecessary CPU cycles.

FWIW, use 640x480@60Hz, it need less filtering. At 320x240 you can use up to 120Hz without image distortion, but there's no benefit to it.

sthalik, 2017-10-16:
I'm closing this issue since this has been discussed a lot.

closed 2017-10-16 by sthalik

rinsatomi, 2017-10-16:
Good show. Thank you!