opentrack/opentrack #693: donation option -- comments?

sthalik, 2017-10-29:
It appears that donations are tax-ignorable in Poland. They aren't meant to be shown on tax forms at all. Quite a few people asked for the ability to donate to the project.

I'd like to know what you feel about it. But first, some ground rules.

- opentrack **WILL NOT** be annoying about the donations (user interface, web site, etc.).
- opentrack **WILL NOT** be obnoxious or desperate about the donation option.
- opentrack **WILL NOT** give preferential treatment to donators. all people's bugs and feature requests are equal, always.
- opentrack **WILL NOT** add new functionality to donators.
- opentrack **WILL NOT** ask for donations by default, only as an option to click when one feels like it. It won't be visible on the main window, and it won't be easy to click accidentally.

I'd like you to express your opinion, especially the against the functionality.

Len62, 2017-10-29:
I think this is a good idea. If people are willing to donate in order to compensate you for providing such a great program, they should be able to do so. The ground rules you set are perfect.

MathijsG, 2017-10-30:
I really like the idea of doing something back to programs I use, either to contribute with money, or with my time, or both. I think it's a sustainable thing helping programs potentially going on.

I think if you do it transparently and without nagging it's perfectly fine for many. So I hope you won't hide the donation button on the website, but just prominently put it somewhere, people still can decide whether they do it or not. I like the Paypal Donation button, because if it's ease of use. Maybe you should dive into that?

Laserclip, 2017-10-31:
I agree. I wrote to sthalik to put the financial support from anyone who wanted to.

There are many people who produce and sell equipment based on their opentrack products and have great financial benefits. And many others who appreciate your work in improving the program.

I am from Poland and here Paypall is not popular so it is good to have other possibilities of donation. Transfer to your account or something else.

Of course, it is better not to focus on large amounts, but it is better to have the opportunity to donate than to not be.

I'm sorry for my English ;)

503Kento, 2017-10-31:
I agree, the ability to donate is a great idea.

MarcinSzczecin, 2017-10-31:
And definitely DO NOT hide this button

Emton, 2017-11-02:
I think it's a good idea opentrack is a great program and you've made some solid updates for it

closed 2018-08-04 by sthalik