opentrack/opentrack #694: [Idea] centering of inertial trackers in fusion tracker configuration

Maetharin, 2017-11-02:
Hello there!
Thx for all the time you put into this and also how you patiently answer our questions.

I usually use the Fusion tracker with the Aruco Tracker for positioning and my EDTracker for more detailed rotation. But even though the Magnetometer inside the EDtracker is supposed to eliminate drifting, I still sometimes suffer from it if I play over a longer period of time. (Which is always)

Would it be possible to put a function into the next or any subsequent versions which recenters the rotation input from my EDTracker (or any other inertial tracker) whenever the other tracker rotates into the center position?

I don´t know anything about programming, so I don´t know how complicated this would be. Does Opentrack register the rotation of the position tracker and disregards it?

Thx in advance for your answer

sthalik, 2017-11-03:
That's too complex and specialized to be exposed directly.

I'm halfway done with plugin support however. All the specialized functionality can be embedded in external plugins with their own dialogs.

sthalik, 2017-11-03:
Does Fusion center correctly with the button press?

Maetharin, 2017-11-03:
yes it does centre perfectly

sthalik, 2018-01-04:
Actually it's doable. Which tracker (input method) do you use?

Maetharin, 2018-01-16:
I am using my EDTracker as rotation source and Aruco as Position tracker in the Fusion Tracker. So inertial and camera.

Thank you for all this effort you put into this, this will greatly enhance my experience!

sthalik, 2018-02-25:
The functionality is too complex by itself. I'm closing this until we have plugin support.

closed 2018-02-25 by sthalik