opentrack/opentrack #695: Back to mouse control

MarcinSzczecin, 2017-11-12:
Hello.

Eurotrack simulator 2
Farming simulator 17
War thunder
American truck simulator
Pro cars

In these games when I use the opentrack to control the image after shutting down the opentrack software it is not possible to return to image control with the mouse as it was before the installation.

The only option to return to mouse control is to uninstall the opentrack. Is there a faster, simpler way to go back to controlling the mouse.

Edit a file, move a file or something else
Thank you

sthalik, 2017-11-13:
> The only option to return to mouse control is to uninstall the opentrack. Is there a faster, simpler way to go back to controlling the mouse ?.

There's an option in freetrack protocol settings to disable tracking for newly-started games, while tracking is not active.

Other than that, no. This is a long-standing bug. I haven't removed it because the existing .dll is whitelisted by games' anti-cheat programs.

sthalik, 2018-01-04:
See #706.

closed 2018-01-04 by sthalik