opentrack/opentrack #696: Profiles not switching or starting on game load

Meridiian, 2017-11-14:
Hi,

I have set all the options correctly to switch to a specific profile on game load and to start tracking but the profiles will not switch and tracking does not start automatically.

sthalik, 2017-11-14:
Did you select the checkbox on the tab?

Meridiian, 2017-11-14:
If you mean the checkbox that says "start profiles from game executable names in this list", yes, I did.

Len62, 2017-11-14:
Are you running OT w/admin privileges? Just tried w/my setup and the correct profiles are being loaded.

Meridiian, 2017-11-14:
Yes it was running as admin. I'm still checking various stuff now but as yet I've had no luck. I have to select a profile manually and click start.

Len62, 2017-11-14:
It should only auto-start if one of the linked games is running when you start OT.

Meridiian, 2017-11-14:
I know, but they're not running.

Len62, 2017-11-14:
Do me a favor, start OT and go to the game detection tab and uncheck "Start profiles...". Close OT and then re-open to see if it is still autostarting.

Meridiian, 2017-11-14:
Well it's working now and no idea what was causing this. I was running another piece of software and I had an error message from it telling me that ETS2 was running, which it wasn't. However, that would explain OpenTrack not switching the profile and auto starting with the ETS2 profile all the time.

I check task manager and ETS2 wasn't loaded, not even a left over or hung process. Rebooting has fixed it but I have no idea why the ETS2 process is still being detected long after it's shut down.

Len62, 2017-11-15:
Glad you got it sorted out. Some games that have a launcher exe that calls the game exe can sometimes fail to shut down properly. With games that have a launcher I bypass it and link directly to the game exe file in the OT game settings.

sthalik, 2017-11-15:
Good thing you got it sorted out! I checked too and seems to work on my end.

closed 2017-11-15 by sthalik