opentrack/opentrack #697: Copy of Aruco headers outdated

e4z9, 2017-11-16:
I tried to build the aruco input part of opentrack, and got into confusing errors because the build does not use the headers from the aruco installation, but its own copy, and that is not up to date with aruco 2.0.19.

The build should use the headers from the aruco installation.

Complicating seems to be that for some reason C++17 is enabled when building against aruco, and the aruco headers use exception specifications that are invalid in C++17.

(I was able to compile by copying the aruco headers over and removing the invalid exception specifications.)

sthalik, 2017-11-16:
Use the `opentrack/aruco` repository rather than the original.

e4z9, 2017-11-17:
The [readme](https://github.com/opentrack/opentrack/blob/unstable/README.md) points to https://github.com/rmsalinas/aruco , which in turn states 

> DO not use this repo, which is not updated. Use sourceforge instead https://sourceforge.net/projects/aruco/

e4z9, 2017-11-17:
Pull request fixes the README

Still, what's the reason for the copy of the headers in the opentrack/opentrack repository? To me that makes even less sense if you only support a fork of ArUco anyhow :)

e4z9, 2017-11-19:
Confirmed that the opentrack/aruco fork of ArUco seems to work as well as the official latest.

sthalik, 2017-11-30:
I agree it should be more intuitive.

closed 2017-12-02 by sthalik