opentrack/opentrack #701: IL-2 Battle of Stalingrad 90 degree pitch axis lock?

NoelShourai, 2017-11-24:
I don't know if this an opentrack exclusive issue or not, but in IL-2: Battle of Stalingrad, the pitch axis seems to be limited to 90 degrees upwards and 45 degrees downwards in the default cockpit view; however, in other view modes such as head-unlinked cockpit view (Alt + F1 by default), the full range of motion is accessible.

I am using opentrack 2.3.9. I've also spoken with others who are using different headtracking devices (TrackIR, etc.) and they don't seem to have this issue. I also don't have this issue in other games (Falcon BMS, etc.).

Len62, 2017-11-25:
Make sure that in the mappings window you set max pitch to 180 degrees.

NoelShourai, 2017-11-25:
Max pitch is set to 180 degrees, as stated it works correctly in other view modes in IL-2, including the head unlinked cockpit view; just not the default cockpit view.

[image]

Len62, 2017-11-25:
IIRC you have to set your pitch down to 180 also in order to get full movement. OT has always been like that.

NoelShourai, 2017-11-25:
Pitch down to 90 degrees was done on purpose. Even so, using asymmetrical mapping doesn't change anything.

To clarify, the default cockpit view's pitch up is locked to 90 degrees, and downward pitch is locked to 45 degrees. The head-unlinked cockpit view (Alt + F1) as well as the external cameras have the full 180 degree range of motion, but does not allow for rotation or XYZ panning. I initially thought this was a restriction with the game, but I was told by people using other tracking devices (TrackIR) that this is not the case.

Len62, 2017-11-25:
The issue may have been raised before, but I think it's a limitation of the view system in BoS. I can't comment on other trackers as I have only used OT.  In the freetrack protocol settings have you tried just enabling the TrackIR protocol, as opposed to the freetrack or both options?

sthalik, 2017-11-28:
As long as your "game data" value is +-180, that's what is sent to the game. opentrack has no further control over how the game interprets that.

NoelShourai, 2017-11-29:
Hi again,

I just tried switching between both protocols/FreeTrack/TrackIR and restarting the game every time I switch protocols, no difference from what I can tell.

I am inclined to believe the restriction is with IL-2 in particular, but I have heard from other people who own the game but use different tracking hardware (a TrackIR5, for example) that they do not have this issue even in the regular cockpit view. It's possible that I misunderstood them, I just wanted to confirm if the restriction actually is with IL-2 itself or if there is something OpenTrack may be missing or not accounting for.

albertolg, 2017-11-29:
@NoelShourai the view is restricted in IL-2, you can easily test it just using the numeric keypad to move the view and you will see that vertical view is a little bit more than 90°

sthalik, 2017-11-29:
@NoelShourai 
> but I have heard from other people who own the game but use different tracking hardware (a TrackIR5, for example) 

We use the exact same method for transmitting the data.

closed 2018-01-04 by sthalik