opentrack/opentrack #704: FreeTrack model Cap - z distance changes as I move left and right my head

gb2111, 2017-12-04:
Hi,
As I move head left and right while looking forward, octopus is getting closer (bigger) each time I am on left or right. Is there way it is only left and right exactly as head is?

Thanks,
Greg

Len62, 2017-12-05:
Go into the mapping window and on the Z axis add some deadspace to compensate for the movement. 

[image]

sthalik, 2017-12-05:
I already know how to implement it. The same feature request gets reported under different names. So little time and energy lately. It's similar to PT calibration, except even easier.

gb2111, 2017-12-05:
thanks for reply.
is this a desired behavior ? will it be implemented as fix or alternative setup ?

sthalik, 2017-12-05:
It'll be implemented as calibrating axis interconnect. A covariance matrix or something more indepth.

gb2111, 2017-12-05:
that is so cool :) 
when are you planning to do it ?
i was assuming this is desired behavoir for monitor games.

sthalik, 2017-12-18:
I'm mostly done with calculating axis interconnect. I haven't started any algorithm for decorrelating the data.