opentrack/opentrack #707: Couple of suggestions

Portash, 2017-12-10:
Sorry if these have been brought up previously.

Curves - the ability to turn on/off curves (instead of having to delete points); ability to copy curves from one axis to another; finer editing of curves - current curve editing is too coarse particularly when trying to design a dead-zone.

Precision mode - option to momentarily freeze or slow down tracking while a key is held, helps spotting targets while view is zoomed in.

sthalik, 2017-12-10:
I agree about mapping snap value. At least the translation axis should be snapped to (`.5`, `1`). Are there any more such cases?

As for "precision mode", this is what the "toggle" key does. Is it sufficient for you? If it was changing mapping scale rather than stopping, we'd get an annoying discontinuity on disengaging tracking stop.

As for manipulating mappings in different ways -- the methods you suggested and many more can be thought up. But then we'd get the mapping interface cluttered with individual ways to modify the mappings. I'd say it's a change for the worse.

You can copy opentrack's mappings using a text editor while opentrack is closed. These `@Variant` binary values aren't user-editable, but they can be copied around.

Portash, 2017-12-11:
Thanks for the reply. The toggle key is exactly what is needed but unfortunately it carries inertia with it. A sudden stop might be too extreme and a tiny bit of inertia would be nice but right now it continues to move over a second after pressing the toggle key.

sthalik, 2017-12-18:
What's your filter and filter settings?

MathijsG, 2017-12-22:
I like the idea of easily enabling or disabling certain axes, like FaceTrackNoIR had. Now I indeed remove all the curves from a certain axis when I want to disable an axis for testing. Would be handy to seperate the curve itself from the enabled/disabled status.

sthalik, 2017-12-22:
Then maybe make disabled status cause no output on that axis?

sthalik, 2017-12-22:
Or rather, split into "disabled" and "disabled (possible output)".

MathijsG, 2017-12-29:
What would be the difference if the output isn't used? Why not only 'disabled' which doesn't send any value of the specified axis at all?

sthalik, 2017-12-29:
"Disabled" allows for possible relative translation or pseudo-neck output.

sthalik, 2017-12-29:
And you're right, having two "disabled" options is just confusing without digging into internals.

MathijsG, 2017-12-29:
Yeah I won't see any use case for it. maybe we should just have Disabled and when some exotic usecase pops up in the end, we can always expand this feature, just like the shortcuts have expanded with more options because of people's use case.

sthalik, 2018-01-03:
I added "Disabled" and "Relative translation only" since I personally want Y downward on pitch down. It should be pretty self-explanatory.

sthalik, 2018-01-04:
@Portash I've implemented your first suggestion.

closed 2018-01-04 by sthalik