opentrack/opentrack #708: UI fails to launch when Moonlight is connected

ouoertheo, 2017-12-15:
I found a bug where if I launch opentrack with Moonlight connected from my Android phone, the opentrack UI will not display at all. The process runs in the back, but nothing at all happens on launch.

I have Windows 10. Not sure what info you would like up front, but let me know and I can get it for you.

sthalik, 2018-01-02:
Does the Moonlight app have a component running on the PC?

Download "sysinternals process explorer", run it, then:

- select the opentrack process in the colored list by a single-click (double-click will open properties, we don't need them)
- view -> show lower pane: enable
- view -> lower pane view: DLLs
- expand the lower pane view so it takes most of the screen
- sort the lower pane by "Path"
- screenshot its pages and post the screenshots. I need to see the entire content. Use Page Down to scroll the list by one page, then do another screenshot, etc.

Here's an example how it's gonna look like

[image]

ouoertheo, 2018-01-02:
I noticed that Nvidia injects their streaming libraries, which is what moonlight uses. Let me see if I can wrangle up all the variables on repro and get you some better information.

Sent from Nine<http://www.9folders.com/>

[image]<https://user-images.githubusercontent.com/1896811/34472326-efc439de-ef5f-11e7-914f-4e07d104c95c.png>

-

ouoertheo, 2018-01-20:
[image]
[image]
[image]
[image]

sthalik, 2018-01-21:
It's that `nvgamepadinput.dll` thing that's breaking opentrack 99%.

sthalik, 2018-05-14:
It's now in FAQ, see <<https://github.com/opentrack/opentrack/wiki/common-issues>>.

closed 2018-05-14 by sthalik