opentrack/opentrack #709: Assetto Corsa not working with OpenTrack

MathijsG, 2017-12-20:
Hi,

Today I was playing with Assetto Corsa, however I couldn't get it to work. How can I debug this the best way? Because none of the free programs do work (Opentrack, Trackhat, FacetrackNoIR)

And the game is incorrectly listed as Lock On: Modern Air Combat:
[image]

MathijsG, 2017-12-20:
Woops somehow the default camera wasn't the free looking in cockpit camera :'). Sorry

closed 2017-12-20 by MathijsG

sthalik, 2017-12-20:
Is it still showing with LOMAC as the game title?

MathijsG, 2017-12-20:
Yup, the same for FaceTrack and Trackhat.

sthalik, 2017-12-20:
I need `sgl.dat` and can't get it myself. Can you send me an email please?

sh

MathijsG, 2017-12-20:
Hi, where can I get this file? I can't find any of this in Assetto Corsa folder.

sthalik, 2017-12-20:
Send me an email, I'll explain.

MathijsG, 2017-12-29:
I emailed you the definitions :-).