opentrack/opentrack #710: tracking small dot sticker on head

fabeit, 2017-12-24:
I would like to use opentracker as a mouse to reduce strain on my tendons. I had a single LED setup with free-track, it works great but I get up from my chair a lot, so the wire and the recalibration were not helpful.
I am trying to find a better setup, like SmartNav or Tobii but not so expensive.

I have noticed aruco marker tracker, is there a way to have a small dot or any small object to attach to forehead or to a headset but without any cable?

Thanks!

sthalik, 2017-12-24:
Yes, aruco will work fine. Things to keep in mind:

- the marker must be a rigid planar square
- keep a bit of empty white space around the marker
- the marker must be placed at angle angle with regard the camera. see wiki's aruco page.

fabeit, 2017-12-24:
How big does the square have to be? Keep in mind that I have to wear this on my face.

sthalik, 2017-12-24:
That is, 7 cm PLUS white area around the marker. The camera needs to distinguish the white area.

fabeit, 2017-12-25:
Thanks for the info, I don't think this will work, can you imagine putting that on your forehead?

sthalik, 2017-12-25:
Definitely. I used a piece of cardboard put between my head and headphones, with the marker mounted on it. Am not kidding.

closed 2018-08-03 by sthalik