opentrack/opentrack #711: Decoding UDP receiver output in Java

D3add3d, 2017-12-25:
Hello, I am trying to decode ouput from the UDP receiver in Java, by looking throught the issues I found that the output is 6 doubles so I am doing
```
csvBuilder.append("x,y,z,yaw,pitch,roll\n"); //StringBuilder
for(int i=0; i < 1000; i++) {
    socket.receive(packet);
    byte[] data = packet.getData();
    DoubleBuffer buff = ByteBuffer.wrap(data).asDoubleBuffer();
    for(int j=0; j < 6; j++) {
        output[j] = buff.get(); //output is array of doubles
    }
    csvBuilder.append(output[0]).append(',').append(output[1])....append('\n');
}
```
which reads the buffer 8 bytes at a time and converts them into a double, the only problem is I am getting values like: 6.8964079183E-313, 3.405463842423E-312, 2.72443581826E-312... (values for x from lines 2, 3 and 4 from the resulting CSV file) so apart from the values being so damn small fractions of a number they are also jumping all over the place and the resulting chart is basically a drawing of noise. I tried reading the values as ints, floats, averaging them and nothing worked so my question is what exactly am I doing wrong?

D3add3d, 2017-12-25:
Ooooh... it was the endianness of the data... flipping the data array worked

closed 2017-12-25 by D3add3d

sthalik, 2017-12-25:
FWIW there's no such thing as bigendian floats, even on bigendian machines.