opentrack/opentrack #712: Relative translation off while held feature desapear.

Plut00n, 2017-12-27:
Please put back this feature that was very useful in WWII Simulations.
This feature was present in 2.3.0, and was joystick compatible.
It was very usefull for scanning the sky whithout activate relative translation when you look behind.

Thanks by advance !

2.3.0 shot
[image]

sthalik, 2017-12-27:
Would it solve your problem to disable relative translation completely, but enable dynamic neck?

Plut00n, 2017-12-27:
I don't know, beacause I never tried dynamic neck.
I suppose this feature try to simulate the neck and shoulder movement you do when you look behind.

I used to like the ability to ON/OFF relative translation, beacause, if ON, when I do a quick view behind, the move of my body (IRL) still natural. But when The move is slow (for spotting ennemies for exemple, and stick your head against the canopy, while doing a slow 180° movement), the Relative Translation become an inconvenience. That's why I used this shortcut according to my needs.

I don't know for dynamic neck, it sounds really cool, but perhaps the ability to choose between both options could be great.

Thanks for your reading !

(sorry for my english, not my native language)

Maetharin, 2017-12-28:
I also think this would be a great feature!

I dislike using the dynamic neck feature to replace relative translation, because it either moves my head to the side when I look at 90°, or makes moving my head to the side when looking ahead feel very unnatural. It really depends on how I set the x axis curve.

I recently saw the TrueView feature in TrackIR on my friends PC, and this solution is perfect. It decouples rotation from position, and position change is always applied after rotation. 
I imagine this is very complicated to implement, though I have no idea about coding and whatnot.

So IMO unless you want to further refine relative translation to become more like the TrueView feature, I think adding a "disable relative translation" button ought to work rather well!

sthalik, 2017-12-28:
How is "TrueView" different from relative translation?

sthalik, 2017-12-28:
@Plut00n we can solve it, but there's probably a better solution than the keyboard shortcut, that can satisfy everyone.

Plut00n, 2017-12-28:
The shortcut solution is quite fine for me, but yeah, there is maybe a better solution for everyone.

For slow movements trought the cockipt (scanning the background for spotting planes, look over the cockpit) you need to moove your body on the side, and when you reach 90° translation, the Relative Translation give me back to the center, witch I don't want. If I reach 180°, the RT give me back to the opposite site, witch is worse. That's why this shortcut is really usefull for me.

For quick movements to look in your 6 o'clock, (really usefull in wwii combat sims), Relative translation is ok, beacause the movement is natural when you do it quickly.

Maybe the better solution would be a way to recognise if the movement is fast or slow, and then on/off automatically Relative Tranlation. I don't know how hard is it to code, I have no knowledge in that.

sthalik, 2017-12-29:
@Plut00n @Maetharin TrueView sounds like exactly what we're doing. Rotate first, then change position relative to where we're looking.

If it behaves differently, I need details.

albertolg, 2017-12-29:
relative translation with this parameters is fine for me, I else don't know how works trackir's trueview
[image]

Maetharin, 2017-12-29:
I think in TrackIR the centre of rotation moves with the head, whereas is Opentrack with Relative View, it stays in the absolute centre.

In Opentrack with Relative Translation on, when I rotate my head after I have moved it forwards, it will also be moved into that direction. In TrackIR, when I do the same in the same situation, only my head will turn, without moving into the direction of the rotation.

I don´t know how they did it, but to get these results I´d use 9 axes instead of 6. 

- 3 rotational axes
- absolute movement axes xyz
- relative movement axes xyz

In the centred position, the absolute and relative axes align. When the head is rotated, the relative axes is aligned with the head, whereas the absolute axes stay the same. Movement of the IRL head, will first be translated onto the relative axes, which then results on a position on the absolute axes.

Again, it may be that True View operates completely differently, I have no idea. But at least the results seem to match with my description.
It also has a feature which realigns the absolute and relative axes again, though I have no idea how that exactly works. 

I hope I have adequately illustrated the difference between the two systems, though I have no idea how much work it would entail to implement changes like this.

sthalik, 2017-12-29:
Did you use the calibration feature in the PT or Aruco tracker? It shouldn't cause movement when you rotate your head.

Maetharin, 2017-12-29:
I use the fusion tracker with an EDTracker for rotation and Aruco for position. I have calibrated the Aruco, and it works pretty well even on its own.

The thing I wanted to describe is that when relative translation is on, and _only_ when it is on, position is translated during rotation. This causes the sudden movement from one side to the other when rotating from looking back to forward again, and it also causes the movement of my head when I´m in a forward position and only want to rotate it.

sthalik, 2017-12-29:
> position is translated during rotation

But that's the reason for "relative translation". How can TrueView function differently?

Maetharin, 2017-12-29:
I´ve written my theory on how they do that earlier, though it may be completely wrong, it does align with the result.

In TrackIR when using TrueView, when you move your head to a certain position, and then turn it, your head stays is position. 

I´ll try to illustrate it further by first describing what your real head does, then your virtual head in both TrackIR with TrueView and Opentrack with Relative translation. 

Movement of real head:

1. Move 5 cm forwards
2. Turn 10° to the left
3. Move 5 cm forwards

In TrackIR with Trueview on this will translate into this:

1. Head moves 10 cm forwards
2. Head turns 20° left
3. Head moves 10 cm along the line of sight.

During step 3, when the reative movement of the head is 10 cm, its absolute movements are, if we follow the rules of trigonometry, the head moves 9,3969cm forwards and 3,4202cm to the left.

In Opentrack with Relative Translation, the following happens.

1. Head moves 10 cm forwards.
2. Head turns 20° to the left, but also follows the absolute movement of step 3 in TrackIR
3. Head moves 10 cm along the line of sight.

So in Opentrack, your head moves even when you only rotate, which feels very unnatural during a game.

Again, I´ve no idea how to code, I just want to add my suggestions on how to make Opentrack even better. It already is great as it is.

sthalik, 2017-12-29:
I'll try my hand at it. Your explanation is good, thanks.

However -- what if you return to your sitting position while rotated? Will it go back to center or not?

sthalik, 2018-01-02:
I have an idea of getting Euclidean distance from the existing position, then lerping it back to the origin position as the user returns. Comments?

sthalik, 2018-01-02:
Or probably use max(x, y, z) as the distance measure. We don't wanna go back unless it's commanded.

albertolg, 2018-01-02:
right now, in the relative translation, I need to disable for Y and pitch or I have weird results

sthalik, 2018-01-04:
Maybe we could limit relative translation to high yaw angles? 60 deg or more for that matter.

albertolg, 2018-01-04:
Who have the original trackir can say how it works?

Maetharin, 2018-01-04:
I‘ve described my impressions I got from TrackIR as closely as possible. 

I don‘t think you can get any closer without asking a developer about it, which would be kinda weird.
„Hey could you tell me the sekrit formulas to your success plz?“

Anyway, I like the idea of limiting Relative Translation to high yaw angles, though I don‘t know if 60 deg would be enough.

A perfect solution would be if it were customisable.

Another idea I had, though i don‘t know how much work it would entail, would be to in- or decrease this yaw angle the further you move your head on the Z axis. 

So if your head is 10cm forwards the angle would be 90deg if 60deg were the standard angle, and 30deg in the opposite direction.

Maetharin, 2018-01-04:
In conclusion, I think the yaw angle solution is very elegant.
Though if you want to try your hand at a TrackIR like solution, I think it would be more intuitive for the user.

Concerning the return to the original position, I think TrackIR recenters your head once you’re inside a certain range of ± yaw angle and near the centre of the X axis.

Maetharin, 2018-01-04:
sry for beginning all these new posts, I can‘t edit old ones on my phone.

I just had another idea for the yaw angle solution.

What about 4 overlapping ranges of yaw angles which lock your head into a relative box of movement. 

It may sound stupid without clarification, so please indulge me.

Let’s put these angles at 120 deg respectively, so -60 to 60, 30 to 150, 60 to -60 (even if the actual yawing only goes towards ±180, again, plz indulge me) and -30 to -150. 
Let’s count them clockwise, so Box 1 to 4 in the order I‘ve written them.

In Box 1 movement of your real head towards the monitor is translated into virtual movement along the Z Axis, in Box 2 the same movement is translated into movement along the X Axis, in Box 3 it‘s translated negatively on the Z Axis

If you‘re in Box 1, you have to yaw your virtual head over 60 deg to get into Box 2. To get back into Box 1, your virtual head must now be yawed back to 30 deg. To get into Box 3, it has to be yawed until 150 deg.

sthalik, 2018-01-16:
@Maetharin it's my understanding that user wants reltrans enabled when they either:

1. look downward at the aircraft switches
2. look behind the aircraft

Also, reltrans doesn't cause that kind of direct transformation. It rotates the user's position offset by his current rotation, taking into aspects all of XYZ position offset.

Maetharin, 2018-01-16:
Thx for the clarification.

How do you achieve the rel trans limit? Rotation over a certain angle?
If so, would it be possible for users to customize the angle at which the limit sets in? Or maybe choose between a predetermined set of options?

I don’t know how much extra work this would entail on your end, but customisability greatly enhances user experience

sthalik, 2018-01-17:
Rotation over certain angle is either of:

1. `|yaw| > 135`
2. `pitch < -30`

When you have a reason to customize, let's talk about it.

albertolg, 2018-01-18:
@sthalik it feels unnatural now with this values, please the older was better... for example, with the new values if I'm looking at my left and I move my head ahead in the game it will just move at right and is not natural

sthalik, 2018-01-18:
Please rephrase.

albertolg, 2018-01-18:
in the test build you changed how the relative translation works... so with the new beaviour if I look 90° left or right and I move my head forward I would expect to move forward, but now the camera move to the left or right and not forward or backward, like the relative translation is disabled and is not natural, what I expect is that if I look 90° left and move my head close to the screen (forward in the Z axe) then the camera will move ahead and not right how is now.. imagine you are in your car and you look at your left window... if you move your head forward you expect to look outside the window and not move the head to the windshield... I don't know if is clear

sthalik, 2018-01-18:
It makes sense.

The question is whether the user is interested in moving toward the side of the canopy with reltrans.

Is looking out of the window actually doable in car simulation?

sthalik, 2018-01-18:
Actually why not only enable reltrans why looking very close to forward? Disable if all are true:

- less than 15 deg yaw
- less than 10 deg pitch
- less than 10 def roll

Would that make everyone happy? WW2 aircraft fans and others as well?

albertolg, 2018-01-19:
I only did an example with the car, I only use flight simulators (DCS, IL2 BOS, Cliffs of Dover), about your suggestion I can only test it and tell if is natural or not

sthalik, 2018-01-19:
See today's build. Relative translation has 3 options now. Mapped yaw has to be below 20, pitch 20, roll 7.

sthalik, 2018-01-19:
I can increase the limits if it's too small for you WW2 players.

albertolg, 2018-01-19:
@sthalik , with this build the old beaviour is back, that's good thanks; but ehm... sorry can you explain what should change changing the reltrans from "Enabled" and "Enabled when not aiming" ? I don't see any difference, maybe I'm looking at the wrong thing?

sthalik, 2018-01-21:
@albertolg if you look closely at the WW2 gun cross with "Enabled", and move toward the screen, a minor yaw or pitch movement will throw your view way off. Can you confirm this?

albertolg, 2018-01-24:
yes, I now noticed the difference...

sthalik, 2018-01-24:
@albertolg is it good enough now?

albertolg, 2018-01-25:
for me "Enabled" is fine, in DCS World, need to test other sims but I think will be the same...

Maetharin, 2018-01-25:
I´m not dure wether the current angles are sufficient to work, but I haven´t yet had the opportunity to test it out.
Would it be much work to have sth like a drop down menu where we could choose between several options? f.e. 15°, 20°, 25°, 30°?

sthalik, 2018-01-25:
It's not about the options being too much work. Sometimes "less is more".

sthalik, 2018-01-25:
Maybe we should add something like "about:config" for options?

Maetharin, 2018-01-25:
Considering that there are not only different kinds of Cockpits, but also different kinds of vehicles you can use OT with, I think having sth like 3-4 options to choose from would be optimal.

I f.e. have different kinds of angles I watch out from in a BF 109 and a Spitfire.

If it isn‘t too much work, would you consider it?

sthalik, 2018-01-26:
The complaints are limited to people using WW2 flight sims.

It's not as much for watching out, the complaint is about precise aiming at the gun cross position.

I may introduce an `about:config` uTorrent- or Firefox-style menu at some point.

With regard to "less is more": if I went with yours and other people's numerous suggestions, you wouldn't like the result. No one would. I go with *some* of the suggestions.

sthalik, 2018-01-26:
Hey, that `about:config` thing is actually easy to do, given that we can already enumerate all the options. I can add some not connected to the UI and they'll show on the list.

biffa66, 2019-06-06:
> relative translation with this parameters is fine for me, I also don't know how works trackir's trueview
> [image]

This works for me too.  Many thanks albertolg https://github.com/albertolg- it was really bothering me

closed 2020-01-27 by sthalik