opentrack/opentrack #713: Problem with realsense 2016 R2 SDK & Windows 10 Creators Update ++

JayMonkey, 2017-12-28:
Hi sthalik,

There is now a known issue with the realsense 2016 R2 SDK and Windows 10 Creators & Fall Updates, essentially its now impossible to install the R2 SDK. There are many reports of this issue on the web EG:

https://software.intel.com/en-us/node/635914

I believe that the issue is related to to fact that since the Creators update Windows 10 now includes support for realsense cameras via a MS driver to support the Windows Hello feature ... sigh .. Thanks M$ for screwing things up .... again.

I've seen this issue my-self and its a pain ...

As such its impossible to now use the current release of Opentrack with realsense hardware on a new/clean install.

However the 2016 R3 SDK HeadTracking feature can still be installed without issue by simply registering with the dev site and installing the Hybrid Face Tracking SDK package which includes the core  and Face Tracking modules.

With the latest release of OpenTrack, when you select realsense as a input device it still prompts to install the R2 SDK even if you have the 2016 R3 SDK already installed.

I understand why you dropped support for the R3 SDK a while back (to allow simple install) but since there is now an issue with R2 it causes a bit of an issue for users.

I personally got round it by using an old non release version you sent me a year or so :- 
opentrack-test-20170219_1-realsense-r3-tmp-do-not-use

However this is based on a quite an old release and not an option for current/new users.

Is it possible for you to add support to opentrack to either automatically detect that the 2016 R3 SDK is already installed or add a setting that can be enabled in the preferences to by-pass the request to install the 2016 R2 SDK ... 

I am Happy to provide testing and feedback if you need it
Appreciate all that you do with keeping the OpenTrack Development moving forward

Cheers
Jay

ph0b, 2017-12-28:
Hi,

previous issue for more background: https://github.com/opentrack/opentrack/issues/544#issuecomment-280745278 

The report you mention seem to be about the camera driver and not the R2 SDK runtime.
Can you confirm it's not possible to install the 2016 R2 SDK on your system? ie. https://github.com/opentrack/opentrack/blob/unstable/tracker-rs/redist/intel_rs_sdk_runtime_websetup_10.0.26.0396.exe?raw=true ?

I can't test right now as my system is stuck on RS1 for now. I can do more testing in 10 days on a Windows RS3 system, to verify if we indeed should move to R3 with all that implies.

sthalik, 2017-12-29:
@ph0b the installer fails to proceed due to "integrity checking failed". It appears remote files aren't there anymore, given that it takes no time to process them.

I have a working SDK installation directory, but obviously will only share it if you and Intel don't mind.

sthalik, 2017-12-29:
@JayMonkey it's not a problem porting it, so we're good for all possible scenarios.

JayMonkey, 2017-12-29:
The issue as i understand it is that the MS Camera Driver is not compatible with 2016 R2 SDK but does work with the 2016 R3 SDK as far as i can understand from the many forum posts of the intel dev site forums and my own experiments .... 

It should be possible to force uninstall the MS driver and re install the Intel one and then manually install the SDK but that negates the whole reason for sticking with SDK R2 (for steam-lined install) also due to the way Windows 10 update works its quite likely that MS update will re-install the ms driver at a later date causing god know what ...

Since Intel dropped in house support/development for their own realsense SDK sometime ago, maybe the time has now come to move on to Realsense SDK 2 .. which is open source and in constant active development ..... so you could potentially build your own module to support realsense cameras  which would negate any need for a 3rd party install (as long as base driver is installed) .... it supports Windows, Linux and OSX ..

The only issue with SDK 2 is (as far as i can tell) only supports realsense based on production chipset's such as SR300 and D400 used in commercially released products (SR300 used Razor Stargazor, D400 used in many new laptops) ...... all older realsense chipsets where only officially released with hardware dev kits ....  

From what i've read Its face tracking is far superior to that supplied by SDK R2 & R3

One possible way forward would be to keep legacy support for older realsense hardware but on the understanding that its down to the user to install the driver and SDK modules.

And have integrated support using your own module for newer realsense hardware using the new opensource SDK 2

Many new laptops, all-in-ones and even some monitors are now shipping with built in realsense hardware using either SR300 or D400 chipset.

Git Repo for SDK 2 is here :- 
https://github.com/IntelRealSense/librealsense/tree/development

Happy to help out in what ever way i can
Cheers
Jay

sthalik, 2017-12-29:
@ph0b what's our agenda then?

ph0b, 2018-01-03:
> @ph0b the installer fails to proceed due to "integrity checking failed". It appears remote files aren't there anymore, given that it takes no time to process them.

The one containing "lantern" in its name is missing, but it doesn't prevent the rest of the R2 runtime to get installed and run.

> From what i've read Its face tracking is far superior to that supplied by SDK R2 & R3

I'd be up for it then, but I'm afraid face tracking isn't part of librealsense. It seems to be a separate closed-source module part of RealSense Linux SDK and not available on Windows, I could only find packages of `librealsense-persontracking-dev` for Ubuntu.

I've just done this test on a clean install of Windows RS3, Fall Creators Update:
- installed 2016 R2 runtime
- plugged RealSense SR300 camera in the system for the first time
- waited for default drivers to get installed (SR300 Depth: 3.3.27.5718)
- run opentrack with RealSense

and it worked flawlessly. 

@JayMonkey maybe your issue is more specific to your system and comes from something else than a complete runtime/DCM incompatibility? In any case I need more information in order to help you/other users who could face the same issues.

Can you try again with default drivers and the 2016 runtime linked above?
If it still doesn't work, please report your system version (RS2, RS3?), the exact camera hardware you're using (Razer Stargazer? Intel SR300 devkit?), what RealSense software is installed on your system (runtimes, SDKs), the driver's version as reported by device manager so I can investigate further.

JayMonkey, 2018-01-03:
@ph0b,

I missed the fact that the SDK2/librealsense face tracking module s currently closed source and only running on linux ...... my bad

That said, since it looks like Intel have dropped developing their in-house realsense SDK's I doubt it wont be too long before its been ported to windows and macos, hopefully at some point it too will be made open source .. so might be worth keeping an eye on it ...

I'll uninstall R3 SDK and go back to a clean system, then try manually installing R2 from the source you linked and let you know if it worked ok ..

Even if it does work it looks like their will be a need to modify the opentrack realsense dependance checking ... now that the official intel R2 web installer no longer works.

I guess you could change the download link to the archived version you have but not sure if you would be crossing any intel license rules ??.

Still would be nice if you could have an option to support SDK R3 (if already pre-installed) ... R3 makes use of some later GEN CPU features it does (IMHO) offer a slight better tracking smoothness as i wrote some time ago:-

https://github.com/opentrack/opentrack/issues/544#issuecomment-282567176

Will test R2 using the method your detailed above and get back to you.

Cheers
Jay

ph0b, 2018-01-03:
To be clear, using R2 from the source I've linked is exactly the same than launching the 2016 R2 install from opentrack itself. It's using the official 2016 R2 *runtime redistributable* web installer and it does still work as stated above while being compliant with the SDK license. As long as this method is working, opentrack deps are good as-is and 2016 R2 is the required dependency, and I hope we'll find a way to have it easily running on your system too.

There is no web based runtime installer for R3 and the offline one is too big to be part of opentrack, so I'd rather skip it unless 2016 R2 can't be used anymore and there is no other solutions.
As soon as there is some open-source or at least cross-platform and usable face tracking module working with librealsense, it'll be definitely better to switch to it.

JayMonkey, 2018-01-04:
I Tried install SDK2 using your link to the web install but it still fails to install SDK2
I only selected the core and head tracking modules 
See attached image for error but its the the same as before and reported by many users ?

Cheers
Jay

[image]

ph0b, 2018-01-04:
I didn't get this issue yesterday.
The downloader is known to be a bit flaky and I've seen reports of users unable to make it work on unreliable connections, is yours supposed to be good? I know it's a pain but did you try more than once and is the download failing instantly or not?

JayMonkey, 2018-01-04:
@ph0b,

Nothing wrong with my internet connection .... high speed and never had an issue with downloading anything else ... web installer runs and performs initial downloads with no problem to get to module selection screen.

Problem occurs after selecting modules to install and continuing to next screen ....

Have tried many times already but always the same issue ... also does it if i try to install SDK 2 on my laptop .... Too many other users reporting the same problem for it to be a end user PC issue..

I wonder if its a server local issue .... ie i'm in the Uk and may get connected to a different server hosting the repo to say someone in the US ?

Since intel are no longer supporting their in house realsense sdk's its unlikely that they will resolve the problem if it is indeed a missing or corrupted file(s) on a server ....

Like i said R3 installs with no issues ...

Cheers
Jay

ph0b, 2018-01-04:
I've tried it today again and now get the same result than yours. I've traced down what the downloader was trying to do, using Wireshark, and indeed the file is missing from the server when I try to download it.
I'm on a different network than yesterday so maybe it's a server sync issue, or a weird coincidence.
I'll report the problem internally, hopefully someone will be willing to fix it.

JayMonkey, 2018-01-04:
Ok .. that makes much more sense now ... could not understand why it worked for you but no one else ...

Cheers
Jay

sir-maniac, 2018-06-03:
Hi.  I've been looking at this problem a bit, and I noticed the runtime DLLs are redistributable, and you can specify a folder to access them when first initilizing the framework.  

Would it be possible to simply include DLLS themselves?  Perhaps in an exe that requires agreement to the EULA before extracting into the opentrack install folder.

efalcon, 2018-12-10:
I am planning to purchase a new Realsense module.
But I found that the new D400 series only support SDK 2.0.
Is it possible to upgrade Opentrack plugin to support RS SDK 2.0?
Thanks.

efalcon, 2018-12-11:
Sorry, I am talking about new SDK 2.0, not SDK 2016 R2 or R3.
R2 and R3 are already end of life.

https://www.intel.com.au/content/www/au/en/support/articles/000026448/emerging-technologies/intel-realsense-technology.html

sthalik, 2018-12-11:
@ph0b can you share your opinion on https://github.com/opentrack/opentrack/issues/713#issuecomment-446102650 please?

sthalik, 2018-12-11:
@efalcon I can make you a build using the old 3.0 SDK. Will that work with your device?

efalcon, 2018-12-11:
> @efalcon I can make you a build using the old 3.0 SDK. Will that work with your device?
I think nope.
The document says new Realsense D400 series only support SDK 2.0.

sthalik, 2018-12-11:
Intel removed the face tracking functionality from the core SDK. There are some third-party packages for face tracking but that requires a decent amount of work to integrate into opentrack. It makes sense to implement all that, and I'll get to it in due time. It's more a matter of months rather than weeks though.

For now, please consider using the PT or [Aruco](https://github.com/opentrack/opentrack/wiki/Aruco-tracker) tracker.

efalcon, 2018-12-14:
> Intel removed the face tracking functionality from the core SDK. There are some third-party packages for face tracking but that requires a decent amount of work to integrate into opentrack. It makes [...]

Oh, I see. Thanks mate.
I will have a try on other devices or just buy an old version (SR300).

efalcon, 2019-01-03:
> @efalcon I can make you a build using the old 3.0 SDK. Will that work with your device?

Hi,
I just bought a SR300 module.
As the old SDK R2 is no longer downloadable, could you please make me the build with R3 support?
I checked the link of Dropbox, the TMP_R3 file has been deleted.
Thanks mate.

sthalik, 2019-01-03:
Does [this work](https://github.com/opentrack/opentrack/issues/544#issuecomment-278932307) for the old SDK?

sthalik, 2019-01-04:
See: https://software.intel.com/en-us/realsense-sdk-windows-eol

Scroll to the botton for "Full SDK". I think it's only necessary to install the core and face tracking components.

sthalik, 2019-03-15:
Please speak up as for whether you need the R2 or R3 sdk, each with their own limitations. Are there any users here still using R2-only devices?

JayMonkey, 2019-03-15:
I'd still like to push for r3 if possible ....

sthalik, 2019-03-18:
It's unfortunate that the driver is before behind Intel's registration-wall, and it's illegal to distribute it otherwise. Adopting the new SDK version is easy but I'd like to wait some more days in case someone speaks up.

sthalik, 2019-03-29:
Okay it's settled for R3.

vaclavdub, 2020-03-12:
Hi,
there is no working solution for d415 or am I missing something? In #924 you confirmed support for d415/d435 but I dont see a way to make it work.
When using SDK R2, I get error: -3.
When using SDK R3, it is asking me to install R2.
New SDK 2.31 doesnt work as well. It keeps asking for SDK R2.
Using opentrack 2.3.11

sthalik, 2020-03-29:
Where'd you get R3 available?

vaclavdub, 2020-03-29:
Hi,

https://software.intel.com/en-us/articles/realsense-sdk-windows-eol

"Downloads" section -> 2016 R3 Individual components - Essentials, Face Tracking and Recognition

sthalik, 2020-03-29:
Please see whether today's test build on <<http://opentrack.misaki.pl>> works.

vaclavdub, 2020-03-29:
Getting "RealSense Tracking Error", Status code: -3.

Using: 
Intel RealSense D400 Series UWP Driver v6.1.160.14
intel_rs_sdk_mw_core_offline_11.0.27.1384
intel_rs_sdk_mw_face_hybrid_11.0.28.2787

sthalik, 2020-03-29:
Can you get the camera working with any other software supporting RealSense?

vaclavdub, 2020-03-29:
Intel Realsense Viewer 2.33.1
RGB Camera and Stereo Module both work

sthalik, 2020-03-29:
According to <<https://software.intel.com/en-us/articles/realsense-sdk-windows-eol>> the D400 camera isn't supported.

2016 R3 Technical Specifications

```
Required Operating System
   F200: Windows* 8.1 or 10 64 bit
   SR300: Windows® 10 64 bit
Supported Intel® RealSense™ Cameras
   SR300
   F200
```

vaclavdub, 2020-03-29:
https://support.intelrealsense.com/hc/en-us/community/posts/360036346553-2016-RealSense-SDK-for-Windows-page-updated

Please note that the 2016 SDK does not have support for the 400 Series cameras

vaclavdub, 2020-03-29:
Only option to make it work with new realsense cameras is to use SDK 2.0, but it doesn't have face recognition implemented. Is this correct? Is there free face recognition implementation somewhere?

sthalik, 2020-03-29:
That is correct. The new SDK is very barebones -- otherwise I'd already be supporting it.