opentrack/opentrack #714: FreePIE IMU APK & OpenTrack not working

tegabxte, 2017-12-30:
I have connected my phone, and whenever I turn on the FreePIE IMU data sending, it does not track at all, but bounces around and changes Game-Data Yaw, Pitch and Roll randomly leading into the vibrating octopus.
Reference picture:
http://prntscr.com/hubsfm

I've tried this with my Samsung Galaxy S5 and am now trying it with my Samsung Galaxy S6 but it does not work. What can I do to fix this?

sthalik, 2018-01-01:
The picture isn't enough. Please send a short video or an animated gif.

ddrmorais, 2018-01-16:
Hi there,

        I use a virtual glass with android 5.5 or lower and the communication between Freepie and Opentrack works well.  But when I try to use my phone with Android 7, I cannot visualize anything in the opentrack display.

sthalik, 2018-01-19:
Try disabling the firewall for the testing duration. opentrack may not have an exception. Do you send orientation or raw data too?

closed 2018-05-14 by sthalik