opentrack/opentrack #716: Link head-pitch to head-height?

henry1491, 2018-01-04:
When receiving input from the EdTracker - which only sends pitch, yaw and roll data - is it possible to link my head's pitch to the height of my head - e.g. the Y axis (it might be X or Z...)?
  
The end result would be - I look (pitch) up, and my head goes up - so I can look over the top of the instruments.  I look down, and my head lowers, so I can see all the instruments, and hardly anything outside - which would be super-helpful for x-plane.
 
Apologies if it's already possible to do this - I couldn't see a way.
 
Thanks for this great piece of software - I use it with Elite Dangerous and x-plane - and it's super!

sthalik, 2018-01-06:
See the "dynamic neck" feature.

henry1491, 2018-01-06:
Will do - many thanks!

closed 2018-01-16 by sthalik