opentrack/opentrack #718: tracker-pt: WII remote: could you consider camera directly outputs points?

cpuwolf, 2018-01-08:
hi sthalik,
could you consider abstract Camera class that has the ability direct output 2D points, such as WII remote?
if so, you can put point::extract as part  job of class Camera. then we can  add WII remote supporting here.

actually I have duplicate tracker-pt code to support WII remote, it works good. but duplicate is not a good idea. hope you can consider supporting WII remote

sthalik, 2018-01-10:
What do you want to replace in `tracker/pt` in total? I can make it into a template class, just make a list.

cpuwolf, 2018-01-10:
hi Sthalik,
I would like to instance a special for example WIImoteCamera from Class Camera. then I can almost reuse whole tracker_pt procedure.  so please abstract Class Camera has the ability direct output 2D points.

cpuwolf, 2018-01-11:
hi Sthalik,
as I look at code at 2.9.3, 'Class Camera' is 'struct Camera' now.

albertolg, 2018-01-11:
@cpuwolf , how do you connect the wiimote? bluetooth? isn't there an adapter or a DIY interface to connect with the usb?

sthalik, 2018-01-11:
I'm abstracting `Camera` and `PointExtractor` then. :+1:!

sthalik, 2018-01-11:
@cpuwolf the API is in `pt-api.hpp` and you can see how to make a module in `module.cpp`.

cpuwolf, 2018-01-11:
@sthalik I never think you are so fast👍. Thank you ver much. let me learn it

sthalik, 2018-01-12:
@cpuwolf please contact me here or via email if you have any questions. Don't worry, I'm a friendly person :smile:

cpuwolf, 2018-01-12:
@sthalik I inhert WIICamera and WIIPointExtractor in below link.

https://github.com/cpuwolf/opentrack/commits/pt-reuse

but I found WIImote doesn't work baseing on cv::Mat& frame. 
this special camera WIICamera should direct output std::vector<vec2>& points,  which I cannot do it in WIICamera.

then look like I have to make WIICamera and WIIPointExtractor they communicate with each other. sounds strange.

sthalik, 2018-01-12:
@cpuwolf,

Currently `cv::Mat` is used unconditionally for image preview. But it makes sense, we shouldn't require PT to use OpenCV's internal format for all camera types. I'm making the change.

There's a class doing the preview, but you can either submit the preview yourself, or do nothing, in which case an empty image will be shown.

> BTW: my cmake+MSVC compile error with your latest baseline, they work OK on v2.3.9.

Please install Visual Studio 2017 Preview. Versions 2017 and 2015 are missing very important C++17 features.

cpuwolf, 2018-01-13:
@sthalik I plan to do migration by 2 steps. 
1. migrate WIImote point tracker from 2.3.1_fix2 to 2.3.9.
2. adapt to latest baseline

but on step 1, I am stuck now. I am confused about pointtracker:POSIT input parameter-points.
per my understanding the input points coordinate, x(-0.5~0.5) ,y(-0.5~0.5). this is so-called screen coordinate as you mentioned at

https://github.com/opentrack/opentrack/commit/537a88e522314753149e1ea7921489e58043b40c

but it does't work on v2.3.9, but i does work on v2.3.1_fix2. again I am comparing POSIT() code between v2.3.1_fix2 and 2.3.9. seems no big change.

if you could help "Ding" me, please comment. thank you. I will continue studying and searching

sthalik, 2018-01-13:
@cpuwolf,

The `Y` equation in `vec2 dt` should be `-(dot.RawY - 768.f/2.f)/1024.f`

These are correct screen coordinates, providing `RawX` and `RawY` are on the `0->1024` and `0->768` intervals. The values for POSIT (screen coordinates) are meant to be on the `(-.5)->+.5` interval. Because the Y image plane has a lower aspect ratio, it can only reach up to 0.375.

For focal length you can borrow the formula computing it from the FOV. Or just return `1`.

I wish we could use templates and get rid of this inheritance API, but Qt doesn't allow for template classes at all.

cpuwolf, 2018-01-13:
@sthalik I don't know how should I thank you. you save time lots of debug time. you don't have donation ha.

[image]

it works now. actually creen coordinate is equal. the root cause is focal length. as you said "return 1" is the "Ding" on my brain. Thank you so much.:+1:. :+1:.

then I can move on to the next step. cheers

cpuwolf, 2018-01-13:
@sthalik I dont know whether I should borther you. I tried vs2017 community preview. compile errors are same
`
“<function-style-cast>”: 无法从“initializer list”转换为“QLatin1String”
`

sthalik, 2018-01-13:
What Qt version are you using? Do you get other errors outside from `QLatin1String`?

Don't worry just ask the questions.

sthalik, 2018-01-14:
I just remembered at the fridge. In `cmake-gui` add `-std:c++17` as another argument to `CMAKE_CXX_FLAGS`.

sthalik, 2018-01-14:
@cpuwolf 

Find the installation location for the preview. It's always in the `Preview` subdirectory. For me it's `C:\Program Files (x86)\Microsoft Visual Studio\Preview\Professional`.

This for me but you can easily adapt:

`C:\Program Files (x86)\Microsoft Visual Studio\Preview\Professional\VC\Auxiliary\Build\vcvarsall.bat`

Open `cmd.exe`, then drag `vcvarsall.bat` to the window. It'll copy the file's full path into the console window. Then add parameters, like this:

> C:\Users\Administrator>"C:\Long\Path\To\vcvarsall.bat" x86

Launch `cmake-gui` from it. Create a new build directory for opentrack and point to the sources. **Use NMake Makefiles** as the generator!

Optionally you can select `opentrack/cmake/msvc.cmake` when CMake asks for a "toolchain file" when generating the build directory for the first time. It sets aggressive optimizations.

While still in the console window, go into the build directory and type `nmake`. It should start building.

Please contact me if you have any problems or need additional information.

sthalik, 2018-01-14:
As long as you're able to get `vcvarsall.bat` work, you can type `cl` and paste me the version number on the first line. Thanks.

sthalik, 2018-01-14:
It should work. opentrack forces `std:c++latest`. I recommend Qt 5.10 or 5.9.x at least.

cpuwolf, 2018-01-14:
@sthalik I used Use NMake Makefiles as generator, then I got exactly same error in console

`
f:\works\github\opentrack\compat\string-literal-operator.hpp(8): error C2440: “
<function-style-cast>”: 无法从“initializer list”转换为“QLatin1String”
`

I guess one possibility is you are using professional version, I am using community version?

sthalik, 2018-01-14:
No, it's not a problem. Can you please switch to English (for example on a different user, or a VM) and show me the compiler output?

cpuwolf, 2018-01-14:
@sthalik hahah. good news! not VS. not preview. it is caused by Qt5.7, after I updated to Qt 5.10. compile passed

cpuwolf, 2018-01-16:
@sthalik I roughly have done the integration. then I cannot test it

one question about module.cpp. should I duplicate one? then how user select different modules? I don't see any UI change

cpuwolf, 2018-01-16:
@sthalik below logic seems not quite correct. why not close camera firstly, then wait thread? because close signal has not chance to send out because thread is blocked by wait()
`
Tracker_PT::~Tracker_PT()
{
    requestInterruption();
    wait();

    QMutexLocker l(&camera_mtx);
    camera->stop();
}
`

sthalik, 2018-01-16:
@cpuwolf  camera is still in use while the `run()` loop runs. It's not meant to be stopped unless `run()` exits.

1. Don't, it's done by the aforementioned `run()` method.
2. Why do you need it? Can't you put in the `stop()` method?

cpuwolf, 2018-01-16:
@sthalik I have block function inside camera

sthalik, 2018-01-17:
@cpuwolf you can wait for it to unblock in `pt_camera` destructor.

cpuwolf, 2018-01-17:
@sthalik non-block re-construction has been done. 

then how user can dynamically select different modules?

sthalik, 2018-01-17:
@cpuwolf link to `opentrack-tracker-pt` in the `CMakeLists.txt` file, initialize `Tracker_PT` in your tracker like in `tracker-pt/module.cpp` subclassing of `pt_runtime_traits`.

cpuwolf, 2018-01-18:
OMG. my understanding was wrong. thank you for your clarification. will integrate and rebase and test, then pull request

cpuwolf, 2018-01-18:
@sthalik I got below link error, seems that one .dll cannot include 2 modules
```
>wii_module.cpp
1>wii_module.obj : error LNK2005: _GetConstructor already defined in module.obj
1>wii_module.obj : error LNK2005: _GetDialog already defined in module.obj
1>wii_module.obj : error LNK2005: _GetMetadata already defined in module.obj
```

sthalik, 2018-01-18:
You can now. Link against `opentrack-tracker-pt-base`. 🐱

cpuwolf, 2018-01-18:
@sthalik OK. common code generates 2 dlls. but see below error
```
1>   Creating library F:/works/GitHub/output/opentrack/tracker-pt/Release/opentrack-tracker-pt.lib and object F:/works/GitHub/output/opentrack/tracker-pt/Release/opentrack-tracker-pt.exp
1>wii_camera.obj : error LNK2019: unresolved external symbol "public: __thiscall wiimote::wiimote(void)" (??0wiimote@@QAE@XZ) referenced in function "public: virtual enum pt_camera_open_status __thiscall pt_module::WIICamera::start(int,int,int,int)" (?start@WIICamera@pt_module@@UAE?AW4pt_camera_open_status@@HHHH@Z)
1>wii_camera.obj : error LNK2019: unresolved external symbol "public: bool __thiscall wiimote::Connect(unsigned int,bool)" (?Connect@wiimote@@QAE_NI_N@Z) referenced in function "private: enum pt_module::wii_camera_status __thiscall pt_module::WIICamera::_get_frame(class cv::Mat &)" (?_get_frame@WIICamera@pt_module@@AAE?AW4wii_camera_status@2@AAVMat@cv@@@Z)
```
if I put my code in same folder, opentrack-tracker-pt will compile with my code, that is unneccessary. so should we re-org the directory structure?
1. base(common code)---static lib
2. pt(folder )opentrack-tracker-pt---.dll
3. wii(folder )opentrack-tracker-wii-pt---.dll
creates 3 folders to avoid unneccessary comple&link

sthalik, 2018-01-18:
Hey,

1. Update to the latest opentrack revision
2. Put your own code in `tracker-wii/` with a short `CMakeLists.txt`. See below for a possible example.
2. Don't create a static library. Link against the static `opentrack-tracker-pt-base` in the latest revision.

Example `CMakeLists.txt`:

```
find_package(OpenCV 3.0 QUIET)
if(OpenCV_FOUND)
  otr_module(tracker-wii)
  target_link_libraries(opentrack-tracker-wii opentrack-tracker-pt-base)
endif()
```

This should work

cpuwolf, 2018-01-18:
@sthalik 🐱  will follow up. Thanks

cpuwolf, 2018-01-19:
@sthalik this time the whole structure is clear, rebase is done.🐱

 sorry but new problem comes up. have you noticed that your tracker-pt has lost its resources?
[image]

sthalik, 2018-01-19:
Yesterday I changed it a bit because the default camera implementation was in `pt-base` by accident. Now it's in the separate module.

Can you create your own dialog, removing things you don't need? If it's a problem don't do it! We could move the original dialog to the `pt` module and have cleaner `pt-base`. But don't waste time if you're ok with the existing dialog.

sthalik, 2018-01-19:
If you `git pull` then the resources are back.

sthalik, 2018-01-19:
Good catch! I was about to release a version with broken images :)

cpuwolf, 2018-01-19:
@sthalik I  would love to reuse UI. the only thing may can improve on UI is "Device", this may confus users. if "Device" can show "WIImote IR camera", that will be better
[image]

sthalik, 2018-01-20:
@cpuwolf amazing work in time for the 2.4.0 release!!! ❤️ 

Which other options from the PT menu you want removed or changed? Camera settings button comes to mind.

sthalik, 2018-01-20:
You also say the FOV is 42 degrees. You also only need the fov in case of establishing the focal length. I recommend you reuse our focal length code, do you want to?

cpuwolf, 2018-01-20:
@sthalik happy hours coorperating with you:heart: cheers!!

Oh. exactly. 
1.  Camera settings is useless for WII
2.  I did some study, FOV of WII is about 42 degrees

but these items is not quite urgent, we can optimize them step by step

cpuwolf, 2018-01-21:
@sthalik 
```
struct OTR_PT_EXPORT pt_camera_info final
```

pt_camera_info is final class, I cannot reuse it

sthalik, 2018-01-21:
A short partial review. Please note the following:

```
	m_pDev = new wiimote;
	if (m_pDev == NULL)
	{
		stop();
		return cam_open_error;
	}
```

`m_pDev` never returns NULL. It's always a pointer to something. That block will never execute. If you can check whether it works change the check expression, or just assume it worked.

I ***heavily*** recommend using

```
#include <memory>

std::unique_ptr<wiimote> m_pDev;
```

Then you can force-destruct using `m_pDev = nullptr;` and create using `m_pDev = std::make_unique<wiimote>("optional-argument", "some-other-optional-argument", ........);`.

This has advantages

- the pointer will not leak when destructing the class
- setting to `nullptr` won't crash if it's NULL already
- setting to new value will free behave as set to `nullptr` first, i.e. won't leak

For focal length computation, in

```
pt_camera_open_status WIICamera::start(int idx, int fps, int res_x, int res_y)
```

Do this:

```
cam_info.fov = 42;
cam_desired.fov = 42;
```

You can set other parameters and they're gonna show as frame size and FPS in the dialog. I'm going to explain how you can reuse the dialog in another post.

The WiiYourself library has a very bad, incorrect license. Written with good intentions but bad as a legal document (which it is). I'm adding the license terms to the `3rdparty-notices/` directory for now. We're going to have the phrase the license requires, added to the `tracker-wii` options so we fully abide by the license terms (as we do with other software we depend on).

Please change the copyright so it points only to you in `tracker-wii`. There's no significant code by others so you're the only author. You can remove me and Patrick since nothing really comes from us.

sthalik, 2018-01-21:
Also @cpuwolf please add the license terms where you added your copyright. It's that "Permission to use..." thing. By default copyright gives no rights to anyone but the author.

cpuwolf, 2018-01-21:
@sthalik How should I thank you for the big reply, I learnt a lot. actually I am not good at cplusplus. thanks for your patient.

https://github.com/opentrack/opentrack/pull/729

# focal length
OK. will update once you are ready

# drawing
I think

[image]

sthalik, 2018-01-25:
@cpuwolf,

#### Focal length

It should be enough to set `fov` to `42` when returning `pt_camera_info`. The code should return a different Z value. Then, remove the calibration value and measure the real distance to the "main" dot that's now the yellow cross. They distances should match in centimeters, as long as the model's values are in millimeters.

Depending on whether the displayed distance and your measurement match, we'd have to decide what to do next.

#### User interface

You can remove individual items from the settings. Open the `.ui` file in PT and look up their names. If they have a generic name like "combobox" or "label", assign proper names and then access them like `ui.something->hide()`. Send the changes to PT's `.ui` file along with the rest.

I've made the change so you can subclass `TrackerDialog_PT` and do changes in resulting class' constructor. I recommend you hide the entire camera settings groupbox since these things don't work for the Wiimote tracker at all.

Don't subclass Qt classes separately when subclassing `TrackerDialog_PT`. It already subclasses `QDialog`, and thus `QWidget`, and `QObject`. Subclassing Qt stuff in a "diamond inheritance" pattern will break the dialog and probably won't compile at all.

#### WiiYourself library

I CC'd you when messaging the library's author. He responded rudely, but also proposed changing his license if someone maintained his original library. We could take it on as an opentrack project, say, `opentrack/wiiyourself`. What do you think?

sthalik, 2018-01-25:
You can also try setting the FOV in the dialog to `42` and see if the distances are real.

cpuwolf, 2018-01-26:
# focal length
now problem is `pt_camera_info::get_focal_length()` is compiled in `libopentrack-tracker-pt-base.lib`
, then while wii has to link with this base lib  `libopentrack-tracker-pt-base.lib`, function `get_focal_length()` is duplicated defined if I redefine this function

# UI
OK. subclassing TrackerDialog_PT

sthalik, 2018-01-26:
1. I was wrong, you don't need to use it directly. Internal PT code uses it.
2. It's `public static` so it's meant to be called directly.

I'm forwarding the guy's rude reply. He wants to change the license for his own code if someone takes care of it.

We COULD move wiiyourself into a separate repo and include it in `tracker-wii` as a submodule. But the question is, do we want even to? It's work and responsibility to maintain it over time. Also I can't test it personally due to lack of a Wiimote.

cpuwolf, 2018-01-26:
haha. I see it. it is really rude.

https://github.com/opentrack/opentrack/pull/735

check my first-step pull request please.

so we have to be maintainer of wiiyourself. I can take it because I have Wiimote in hand.

sthalik, 2018-01-26:
@cpuwolf I'd also like to make some quality changes. Can you work with me on that?

I'd like to grant you developer access. Do you accept? There are only few basic rules. Please send me an email.

cpuwolf, 2018-01-26:
@sthalik yes, I accept. thank you very much. kernel.org has my name, so I know how to cooperate. only think I worry is I am not good at cplusplus, because I am assembly-code guy. 😄

sthalik, 2018-01-26:
@cpuwolf When I started back on facetracknoir I didn't know any C++ at all. After the fork, until C++11 came out, opentrack had real messy code because C++ wasn't any good as a language.

There are few good features in C++ you'll appreciate:

- RAII saves from having to write data-freeing epilogues or labels on each return path, including error paths. It's used the same way for using scoped mutexes.
- The template system is algebraic, and most importantly it can compute like a real programming language, except fully at compile time. People use it to great success in libraries like Eigen.

You'll see that opentrack doesn't use OOP "just because" like some dogma, it's closer to being written in a multiparadigm language. 

- - -

Here are some ideas that may give you the feel of the project. There aren't really any codified "rules". Common sense on how to behave with people is enough to proceed.

- We're using the best solutions there are, no matter who comes up with them. If some piece of code I wrote is bad, you can just say so and discuss it.
- Discuss refactorings of the core code. We don't wanna have conflicting versions. Overall it's better to discuss more rather than less.
- As long as you maintain the tracker and no one else messes with it, but you can mess with "unowned" code. It's only OK for others to change the code you maintain if it fails to compile or work due to some core code change. These changes are normally very minor.
- You **do not** have to be professional like some kind of corporate representative, we're all volunteers here, and users understand this. It's your choice. Just don't be rude toward the users. Act like a normal person does in a group and it's enough.
- opentrack **never** accepts GNU GPL-licensed code. As for GNU Lesser GPL, see `CONTRIBUTING.md`.

Some more ideas.

 - It's OK for internal discussions to take place here, in public. It's better for disagreements to be public, for the sake of transparency. But it's fine to discuss over email all the same.
- The "unstable" branch is marked as protected on Github and doesn't accept rewriting history. It can only be disabled for a split second in case of some fuckup.
- If you know how to use git, try to avoid merge commits and rebase "oops" commits. If you don't know how to rebase, just make merge commits and "oops" commits anyway.

You're being sent an invite to the project.

sthalik, 2018-01-26:
Please note, only (5) is a "hard" rule, because it's the law.

But here's another thing to note (the numbering is a poor joke).

101. Don't expect users to know English perfectly. Don't close issues because of this. I'll personally handle it if it's a problem.

Personally I tend to change my vocabulary and grammar when dealing with people with visibly poor understanding of English. Sometimes making grammar errors on purpose so that the sentence structure is more simple. It's not judgemental, just certain people weren't taught or self-taught English as much.

cpuwolf, 2018-01-26:
some summary in my mind:
1. no GPL, as far as I know GPL some version is like a worm.
2. be nice
3. git I am good at it.

sthalik, 2018-01-26:
Welcome aboard.

cpuwolf, 2018-01-26:
🍺 🍺 

❔  `void set_fov(double value)`  is a setting function, return? 😃

sthalik, 2018-01-26:
Just do nothing in `pt_camera::set_fov()`. When you must return `pt_camera_info` in `pt_camera`, set the `fov` element to `42`. Set `res_x` and `res_y` to real values, i.e. `1024` and `768`. Set `fps` to what you see fit. You can set `idx` to `0` to be safe but it's presently not checked by the code.

[42 more replies]

closed 2019-04-24 by cpuwolf