opentrack/opentrack #719: tracker-pt: when I yaw more than 90 degree, raw yaw data will exceed 90 degree

cpuwolf, 2018-01-09:
when I yaw more than 90 degree, raw yaw data will exceed 90 degree, the only workable recovery is stop opentrack and start again.

I am thinking why POSIT() can produce raw data more than +90 degree, for head tracking, we should never accept these kind of data, maybe we can abandon them somehow.   then fetch previous <90 pose as reference for next iterating.

the only acceptable raw yaw data should be -90<raw yaw < +90.

cpuwolf, 2018-01-09:
pitch has same problem.

sthalik, 2018-01-09:
Select "dynamic pose" in PT options, as this appears to be your issue.

cpuwolf, 2018-01-09:
"dynamic pose" is enabled by default. OK. thanks, anyway I would like to migrate wii mote to latest baseline 2.9.3, and give a try.

closed 2018-01-13 by cpuwolf