opentrack/opentrack #720: make flatpak packages for Linux

sthalik, 2018-01-09:
Making flatpak packages is easier than Debian boilerplate.

Running continuous integration on Linux will help with compile-testing at the very least. Maybe even some very simple unit tests with offscreen rendering.

sthalik, 2018-01-09:
Actually opentrack needs access to the global system due to Wine integration, etc.

jwmullally, 2022-10-13:
Here's a simple Flatpak build for OpenTrack: [com.github.opentrack.test.json.txt](https://github.com/opentrack/opentrack/files/9779722/com.github.opentrack.test.json.txt). I've seen a few people ask for one.

It works well with the neuralnet camera input and libevdev Joystick and UDP Network outputs. It would be good to package something like this up on Flathub even with limited inputs/outputs, especially for container only OS's like Fedora Silverblue, Endless OS, Elementary OS, etc.

Building:
```
flatpak-builder --user --install --force-clean ../com.github.opentrack.test.build com.github.opentrack.test.json
```

Running:
```
sudo modprobe uinput
sudo chgrp wheel /dev/uinput
flatpak run com.github.opentrack.test
```

With `--socket=wayland`, the "Options" window crashes, so I left it out and it works fine with X11 fallback.

---

You can use [jstest-gtk](https://flathub.org/apps/details/io.gitlab.jstest_gtk.jstest_gtk) to see the all 6 output axes (and 1 button).

It works OK with DCS under Flatpak Steam. With Steam Input disabled, it shows up as an "opentrack headpose" joystick with 5 assignable axes that all work (X=X, Y=Y, Z=PITCH, RX=Z, RY=YAW, ROLL was missing for some reason. Seems like Steam/Proton/DCS mapped it down to a simpler joystick or something).

jwmullally, 2022-10-15:
@sthalik : You could re-open this issue here if there is interest in this. It would need a few adjustments to [add to Flathub](https://github.com/flathub/flathub/blob/master/CONTRIBUTING.md).

karuoxa, 2022-10-16:
I did a quick test of the flatpak example made by @jwmullally yesterday and it it worked very well during my test.
It was however limited to PointTracker/Evdev Virtual Joystick for use with the game War Thunder on Arch.

It would be amazing to see an official flatpak of Opentrack.

mathis996, 2025-07-22:
With the end of Windows 10 support and the increasing number of Linux users, it would be really nice if a flatpak version could be created. At the latest with the release of SteamOS for PCs, some will certainly switch over. In order to be able to offer a good transition, it would be just as good if a flatpak version were available. As has already been written, not everything has to work straight away. You can certainly work with the basic functions. 

Is there any possibility of pushing this forward? Unfortunately, I am a complete beginner myself, but I would like to help.

mik-ekoit, 2025-07-22:
I second this. A Flatpak with all the modules enabled, like you get in Windows, would be ideal.