opentrack/opentrack #721: compilation error on archlinux

ljgdasfhk, 2018-01-11:
`[ 77%] Building CXX object proto-libevdev/CMakeFiles/opentrack-proto-libevdev.dir/ftnoir_protocol_libevdev.cpp.o
In file included from /home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.cpp:1:0:
/home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.h: В функции-члене «virtual QString evdev::game_name()»:
/home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.h:28:16: ошибка: нет декларации «otr_tr» в этой области видимости
         return otr_tr("Virtual joystick for Linux");
                ^~~~~~
/home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.h:28:16: замечание: suggested alternative: «strstr»
         return otr_tr("Virtual joystick for Linux");
                ^~~~~~
                strstr
In file included from /home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.h:11:0,
                 from /home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.cpp:1:
/home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.cpp: В функции «IProtocol* GetConstructor()»:
/home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.cpp:97:28: ошибка: invalid new-expression of abstract class type «evdev»
 OPENTRACK_DECLARE_PROTOCOL(evdev, LibevdevControls, evdevDll)
                            ^
/home/sobolev/opentrack/api/plugin-api.hpp:57:20: замечание: in definition of macro «OPENTRACK_DECLARE_PLUGIN_INTERNAL»
         return new ctor_class; \
                    ^~~~~~~~~~
/home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.cpp:97:1: замечание: in expansion of macro «OPENTRACK_DECLARE_PROTOCOL»
 OPENTRACK_DECLARE_PROTOCOL(evdev, LibevdevControls, evdevDll)
 ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.cpp:1:0:
/home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.h:18:7: замечание:   поскольку следующие виртуальные функции являются чистыми в пределах «evdev»:
 class evdev : public IProtocol
       ^~~~~
In file included from /home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.h:11:0,
                 from /home/sobolev/opentrack/proto-libevdev/ftnoir_protocol_libevdev.cpp:1:
/home/sobolev/opentrack/api/plugin-api.hpp:101:27: замечание:   virtual module_status module_status_mixin::initialize()
     virtual module_status initialize() = 0; // where to return from
                           ^~~~~~~~~~
make[2]: *** [proto-libevdev/CMakeFiles/opentrack-proto-libevdev.dir/build.make:76: proto-libevdev/CMakeFiles/opentrack-proto-libevdev.dir/ftnoir_protocol_libevdev.cpp.o] Ошибка 1
make[1]: *** [CMakeFiles/Makefile2:2606: proto-libevdev/CMakeFiles/opentrack-proto-libevdev.dir/all] Ошибка 2
make: *** [Makefile:130: all] Ошибка 2`

ljgdasfhk, 2018-01-13:
[ 55%] Building CXX object gui/CMakeFiles/opentrack-user-interface.dir/keyboard.cpp.o
/home/sobolev/opentrack/gui/keyboard.cpp:33:6: ошибка: «KeyboardListener» не был декларирован
 void KeyboardListener::keyPressEvent(QKeyEvent* event)
      ^~~~~~~~~~~~~~~~
/home/sobolev/opentrack/gui/keyboard.cpp: В функции «void keyPressEvent(QKeyEvent*)»:
/home/sobolev/opentrack/gui/keyboard.cpp:45:9: ошибка: нет декларации «close» в этой области видимости
         close();
         ^~~~~
/home/sobolev/opentrack/gui/keyboard.cpp:45:9: замечание: suggested alternative: «pclose»
         close();
         ^~~~~
         pclose
/home/sobolev/opentrack/gui/keyboard.cpp:49:14: ошибка: нет декларации «key_pressed» в этой области видимости
         emit key_pressed(QKeySequence(event->key() | event->modifiers()));
              ^~~~~~~~~~~
/home/sobolev/opentrack/gui/keyboard.cpp:49:14: замечание: suggested alternative: «keyPressEvent»
         emit key_pressed(QKeySequence(event->key() | event->modifiers()));
              ^~~~~~~~~~~
              keyPressEvent
make[2]: *** [gui/CMakeFiles/opentrack-user-interface.dir/build.make:159: gui/CMakeFiles/opentrack-user-interface.dir/keyboard.cpp.o] Ошибка 1
make[1]: *** [CMakeFiles/Makefile2:1758: gui/CMakeFiles/opentrack-user-interface.dir/all] Ошибка 2
make: *** [Makefile:130: all] Ошибка 2

ljgdasfhk, 2018-01-15:
[ 55%] Building CXX object gui/CMakeFiles/opentrack-user-interface.dir/main-window.cpp.o
/home/sobolev/opentrack/gui/main-window.cpp:47:6: ошибка: «MainWindow» не был декларирован
 void MainWindow::annoy_if_root()
      ^~~~~~~~~~
/home/sobolev/opentrack/gui/main-window.cpp: В функции «void annoy_if_root()»:
/home/sobolev/opentrack/gui/main-window.cpp:52:31: ошибка: некорректное использование «this» в функции, не являющейся элементом
         QMessageBox::critical(this,
                               ^~~~
/home/sobolev/opentrack/gui/main-window.cpp:53:31: ошибка: нет декларации «tr» в этой области видимости
                               tr("Running as root is bad"),
                               ^~
/home/sobolev/opentrack/gui/main-window.cpp:53:31: замечание: suggested alternative: «tm»
                               tr("Running as root is bad"),
                               ^~
                               tm
/home/sobolev/opentrack/gui/main-window.cpp:57:31: ошибка: некорректное использование «this» в функции, не являющейся элементом
         QMessageBox::critical(this,
                               ^~~~
/home/sobolev/opentrack/gui/main-window.cpp:62:31: ошибка: некорректное использование «this» в функции, не являющейся элементом
         QMessageBox::critical(this,
                               ^~~~
make[2]: *** [gui/CMakeFiles/opentrack-user-interface.dir/build.make:183: gui/CMakeFiles/opentrack-user-interface.dir/main-window.cpp.o] Ошибка 1
make[1]: *** [CMakeFiles/Makefile2:1758: gui/CMakeFiles/opentrack-user-interface.dir/all] Ошибка 2
make: *** [Makefile:130: all] Ошибка 2

sthalik, 2018-01-16:
Please run the build as
```
env LC_ALL=C LANG= LANGUAGE= cmake --build .
```

The error messages are meant to be shown in English.

ljgdasfhk, 2018-01-16:
[ 55%] Building CXX object gui/CMakeFiles/opentrack-user-interface.dir/main-window.cpp.o
/home/sobolev/opentrack/gui/main-window.cpp:47:6: error: 'MainWindow' has not been declared
 void MainWindow::annoy_if_root()
      ^~~~~~~~~~
/home/sobolev/opentrack/gui/main-window.cpp: In function 'void annoy_if_root()':
/home/sobolev/opentrack/gui/main-window.cpp:52:31: error: invalid use of 'this' in non-member function
         QMessageBox::critical(this,
                               ^~~~
/home/sobolev/opentrack/gui/main-window.cpp:53:31: error: 'tr' was not declared in this scope
                               tr("Running as root is bad"),
                               ^~
/home/sobolev/opentrack/gui/main-window.cpp:53:31: note: suggested alternative: 'tm'
                               tr("Running as root is bad"),
                               ^~
                               tm
/home/sobolev/opentrack/gui/main-window.cpp:57:31: error: invalid use of 'this' in non-member function
         QMessageBox::critical(this,
                               ^~~~
/home/sobolev/opentrack/gui/main-window.cpp:62:31: error: invalid use of 'this' in non-member function
         QMessageBox::critical(this,
                               ^~~~
make[2]: *** [gui/CMakeFiles/opentrack-user-interface.dir/build.make:183: gui/CMakeFiles/opentrack-user-interface.dir/main-window.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1758: gui/CMakeFiles/opentrack-user-interface.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

ljgdasfhk, 2018-01-16:
[ 57%] Linking CXX shared library libopentrack-user-interface.so
[ 57%] Built target opentrack-user-interface
[ 59%] Generating ui_new_config.h
[ 59%] Generating moc_new_file_dialog.cpp
[ 59%] Generating moc_main-window.cpp
[ 59%] Generating ui_main-window.h
Scanning dependencies of target opentrack-executable
[ 59%] Building CXX object variant/default/CMakeFiles/opentrack-executable.dir/main-window.cpp.o
In file included from /home/sobolev/opentrack/gui/settings.hpp:5:0,
                 from /home/sobolev/opentrack/variant/default/main-window.hpp:13,
                 from /home/sobolev/opentrack/variant/default/main-window.cpp:9:
/home/sobolev/opentrack-build/gui/ui_settings-dialog.h:31:10: fatal error: process_detector.h: No such file or directory
 #include "process_detector.h"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [variant/default/CMakeFiles/opentrack-executable.dir/build.make:79: variant/default/CMakeFiles/opentrack-executable.dir/main-window.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:332: variant/default/CMakeFiles/opentrack-executable.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

akhenakh, 2018-01-16:
Same problem here, does not seem related to Arch.

Had to target a specific commit to make it compile: 

`source=("git+https://github.com/opentrack/opentrack.git#commit=a46f6d07c5f8c6d65940b2d8ec62535ee1983315" "opentrack.desktop")`

sthalik, 2018-01-17:
It works as of right now.

I haven't tested individual trackers, but the changes that broke the Linux build were isolated in the main UI module.

closed 2018-01-17 by sthalik